petercorke/robotics-toolbox-python #1: [Question]

castarco, 2015-04-16:
Hello @petercorke ,

I've cloned an "alternative" repository of *robotics-toolbox-python* from Google Code:
  * https://code.google.com/p/robotics-toolbox-python/

I've seen there's someone with the email ryanno1@gmail who have contributed changes, but his/her changes aren't in this repository. And the "opposite" is also true, your changes since 2008-04-06 aren't in the other repository.

I've cloned it in https://github.com/castarco/robotics-toolbox-python/ , I don't know if it's an interesting idea to merge the contributions of both, or if it's preferable to use your version (I'm new in the robotics world, and I can't properly compare the two codebases).

Thanks for your time.

petercorke, 2015-04-18:
I think this is an error I made in exporting the repo from svn to git.  I will try and redo this.

castarco, 2015-04-19:
Ok, can I help to solve this issue doing something? I prefer to use Python than using Matlab for my robotics practices, but I have to work with other people and I want to make easier for them to work with this library.

I want to push it to the PyPI packages repository ( https://pypi.python.org ), but I don't know which branch I should choose.

I prefer contributing to your project since you are the main contributor and the project manager, but I don't know if this is a discontinued project, and the "branches problem" also affects this point.

Regards.

petercorke, 2015-04-19:
It’d be great to have some help.  This work happened a long time ago, so I’d probably organise the code and the object structure a bit differently now.  It might be a bit premature to push it to PyPI just yet.

I’ve had some issues with the transition from svn to github, and am currently trying to figure out what’s happened there.  That’s a priority for me.

As far as I can tell there is only 1 branch.

peter

Galen-dp, 2015-05-02:
I am willing to help as well. I like what I have seen in your book and the MOOC classes.

Any suggestions on where to start with updating this code leading up to making a PyPI package?

petercorke, 2016-09-19:
Guys, sorry for the stupidly long delay.  I've spent a lot of the last year working on a major revision to the book and the MATLAB code.  I'd really like to move on the Python port.  Any ideas on a good way to automagically translate code?

thomper, 2016-09-19:
I've been looking into it again lately and the MATLAB compiler is all I can come up with. I believe it let's you compile your MATLAB source to a C library. You can then call C functions from Python using the ctypes module. I'm planning to try it out but I have little time I can spend on it right now.

Failing that, I see the author of smop (victorlei on github) started working on it again a few months back. Getting smop to 1.0 might be the next simplest option.

Galen-dp, 2016-09-20:
Glad to see you are looking to update the Python port. Unfortunately I'm not in a good spot to help much right now.

nnadeau, 2017-01-23:
Hey everyone, I just stumbled upon this thread while reviewing some old notes. 

I've been working on a personal Python toolbox, inspired from the original MATLAB concept by @petercorke, for my research and industrial work. Back when I checked for a toolbox Python port, I had assumed the project was abandoned. 

While my project is geared towards machine learning, optimization, calibration, etc of serial robots, I think there'd be a lot to gain if we combined efforts.

PyPi: https://pypi.python.org/pypi/pybotics/
GitHub: https://github.com/nnadeau/pybotics

petercorke, 2017-01-23:
Hi Nicholas, I’d be interested to know more about your project.  My project hasn’t progressed, lots of people say it’s a nice idea but nobody has the cycles to spare.  I’ve been swamped by book edition 2 and RTB 10 (coming soon).

A possibility that intrigues me a lot at the moment is the Julia language, check out http://julialang.org <http://julialang.org/>

It’s scary that it’s only at rev 0.5 but the language features are brilliant.

Peter

nnadeau, 2017-01-24:
> pybotics>
> petercorke/robotics-toolbox-python/issues/1#issuecomment-274533050>, or
> 143418/17495839/a5054eac-5d88-11e6-95fc-7290892c7bb5.png> <https://cloud.
> githubusercontent.com/assets/143418/17495839/a5054eac-5d88-
> 11e6-95fc-7290892c7bb5.png> <https://cloud.githubusercontent.com/assets/
> 143418/15842166/7c72db34-2c0b-11e6-9aed-b52498112777.png> <https://cloud.
> githubusercontent.com/assets/143418/15842166/7c72db34-2c0b-
> 11e6-9aed-b52498112777.png> <https://github.com/
> petercorke/robotics-toolbox-python> <https://github.com/
> petercorke/robotics-toolbox-python> <https://pypi.python.org/pypi/
> pybotics//r/nGitHub:> <https://pypi.python.org/pypi/pybotics//r/nGitHub:>
> python/issues/1#issuecomment-274533050> <https://github.com/

Hi Peter,

Thanks for the interest and best of luck on the new book. For my project,
my particular research domain is precision/medical robotics, so calibration
and adaptive control are big aspects. I felt it was important to interface
with popular packages such as numpy, scipy, scikit-learn, tensorflow, etc.

I've played with Julia a bit, and it's quite fascinating, especially in the
big data and machine learning domain. However, while it is a great tool for
that particular field, the same argument can also be made for languages
such as Go and Rust. It depends on the purpose of the project: do we want
it to be hyper-performant or accessible and widespread in usage?

Personally, I often find code/projects/products to be more efficient and
effective when they use popular languages (C/C++, Java, Python).

santiagoos, 2018-02-21:
Hi, i'm new with the toolbox and python, i had some problems when installing the toolbox, so how is the correct way to install it. Thanks

petercorke, 2018-02-22:
that Python toolbox is no longer supported.  Try robopy instead.

peter

closed 2020-11-06 by jhavl