petercorke/robotics-toolbox-python #2: License Question

Galen-dp, 2015-05-02:
I noticed there is no license listed with this repository. Which license are you releasing this under?

petercorke, 2015-05-03:
Doug,

I was planning on releasing under LGPL again, but willing to be persuaded.

I’m not real happy with the svn to git export, a lot of history has gotten lost on each of the projects, from the point where I made radical changes to the file hiearchy.  I’m looking into ways to fix this before I get to committed to the repos that are currently live.

This is not helped by me floundering about git.

peter

Galen-dp, 2015-05-03:
LGPL sounds good to me.

I am going to try to convert the https://github.com/nirvdrum/svn2git tool.

Git does take a bit of time to learn but it is much easier to use than svn.

closed 2020-11-06 by jhavl