petercorke/robotics-toolbox-python #18: Thanks Professor Corke.

mithi, 2020-04-18:
Thanks for making world-class robotics education accessible. I have compiled all the places online where you can learn robotics in this [README](https://github.com/mithi/robotics-coursework). I put **robotacademy.net.au** at the very top of the list because it's the best in my opinion. Thank you for all of this, your master class inspired me to [build my own robot simulator](https://robotacademy.net.au/). Thanks again.

closed 2020-04-18 by mithi

petercorke, 2020-04-18:
Thanks Mithi,

I’ll link to this from my sites.

There’s a burst of energy again for a Python version of the toolbox, hopefully I can keep the momentum and finish it this time.

peter

mithi, 2020-04-19:
Thanks for taking the time to read my messages and checkout my project.
I hope I can help out in the future if any beginner-level tasks come up
in your repository's list of issues (good first issue label).
Thanks again!