petercorke/robotics-toolbox-python #134: Payload and rne computation

williamledda, 2020-10-07:
It seems there is a bug computing the inverse dynamics with rne method.

Take a look at the following script that compute the inverse dynamics for a Puma560 (but this is the same for each model)

```python
import numpy as np
import roboticstoolbox as rtb

if __name__ == '__main__':
    # load a robot model
    p560 = rtb.models.DH.Puma560()

    p560.q = [
        45 * np.pi / 180,
        70 * np.pi / 180,
        -100 * np.pi / 180,
        60 * np.pi / 180,
        25 * np.pi / 180,
        -140 * np.pi / 180,
    ]

    zero_v = np.zeros(p560.n)

    # p560.payload(m=0, p=np.zeros(3))
    rne = p560.rne(q=p560.q, qd=zero_v, qdd=zero_v)

    tau_str = ''

    for i in range(len(rne)):
        tau_str += '{:.3f} '.format(rne[i])

    print(tau_str)
```

If I run the script without setting any payload, the result that I have is the following

```
-0.000 16.368 4.595 -0.005 -0.004 0.000 
```

If now I now remove the comment to set the payload

```python
p560.payload(m=0, p=np.zeros(3))
```

I have the following result instead.

```
-0.000 16.024 4.381 0.000 0.000 0.000
```

I've reported a similar issue for the MATLAB toolbox, I think that the behaves the same.

williamledda, 2020-10-07:
I think that the problem is that in the payload method of the Dybamics class

```python
    def payload(self, m, p=np.zeros(3)):
        # ...
        lastlink.m = m
        lastlink.r = p
````
So when setting the payload, this will replace the mass of the last link defined in the DH model. Instead of replacing it, the mass should be added instead, or the payload should be considered as 'additional' link. This would actually fix the issue. Not completly about what to do with the the center of mass. It should be added as well?

petercorke, 2020-11-08:
This is on the list of things to fix, it was a quick hack in the MATLAB version and not really proper.

petercorke, 2026-08-25:
Firstly, I'd like to apologise for taking well over five years to get back to you, but essentially my day job and life got in the way. Over the last few months I've been able to devote a lot more time to this toolbox — fixing bugs, reworking dubious design choices, and developing tests and solid CI processes to prevent future drift.

On this specific one: I re-ran your exact repro on current `main` and confirmed it's still broken — `payload(m=0, p=zeros)` measurably changes the RNE result (up to 0.34 difference in torque) despite being supposed to be a no-op. Your original diagnosis ("a quick hack in the MATLAB version and not really proper") still holds. Tracked properly, with the current repro, as #638 — thank you for the original report, and sorry it took this long to get real confirmation rather than just a comment.

closed 2026-08-25 by petercorke