petercorke/robotics-toolbox-python #136: roboticstoolbox has no attribute tools

GRawhideMart, 2020-10-18:
Good morning. I was running the first examples on the README, but, as in title, I arrived at the `qt = rtb.tools.trajectory.jtraj(p560.qz, p560.qr, 50)` part, and the answer was:

```
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
AttributeError: module 'roboticstoolbox' has no attribute 'tools'
```
The system's information is the following:
```
>>> import sys
>>> print(sys.version)
3.8.5 (default, Jul 28 2020, 12:59:40) 
[GCC 9.3.0]
```

petercorke, 2020-11-08:
This is still under rapid development but as of today this works as written.  examples/readme.py is the actual example for the README.  It has changed a bit.  Let us know if this is working for you now.

petercorke, 2020-11-18:
You can use either `rtb.jtraj()` or `rtb.trajectory.jtraj()`

closed 2020-11-18 by petercorke