petercorke/robotics-toolbox-python #160: Getting trouble on running the example code

YijiongLin, 2020-12-14:
Hi,

I'm getting trouble with the code posted here: https://github.com/petercorke/robotics-toolbox-python/blob/master/examples/mmc.py 

```
Traceback (most recent call last):
  File "mmc.py", line 85, in <module>
    c = np.r_[-panda.jacobm().reshape((n,)), np.zeros(6)]
  File "/Users/bourne/Downloads/python_test/robotics-toolbox-python/roboticstoolbox/robot/ERobot.py", line 1113, in jacobm
    manipulability = self.manipulability(
  File "/Users/bourne/Downloads/python_test/robotics-toolbox-python/roboticstoolbox/robot/Robot.py", line 441, in manipulability
    q = getmatrix(q, (None, self.n))
  File "/Users/bourne/opt/anaconda3/envs/rtb_rvc/lib/python3.8/site-packages/spatialmath/base/argcheck.py", line 182, in getmatrix
    raise TypeError('argument must be scalar or ndarray')
TypeError: argument must be scalar or ndarray
```

The error seems to be raised in 

And in the spatialmath code, I found this:

`def getmatrix(m, shape):
    r"""
    Convert argument to 2D array

    :param m: input value
    :param shape: shape of returned matrix
    :type shape: 2-tuple
    :raises ValueError: if ``m`` is inconsistent with ``shape``
    :raises TypeError: if ``m`` is not required type
    :return: a 2D array
    :rtype: NumPy ndarray
    :raises TypeError: if value is not a scalar or Numpy array
    :raises ValueError: if value is not of the specified shape

    ``getmatrix(m, shape)`` is a 2D matrix with shape ``shape`` formed from
    ``m`` which can be a 2D array, 1D array-like or a scalar.

    .. runblock:: pycon

        >>> from spatialmath.base import getmatrix
        >>> import numpy as np
        >>> getmatrix(3, (1,1))
        >>> getmatrix([3,4], (1,2))
        >>> getmatrix([3,4], (2, 1))
        >>> getmatrix([3,4,5,6], (2,2))
        >>> getmatrix(np.r_[3,4,5,6], (2,2))

    .. note::

       - If ``m`` is a 2D array its shape is compared to ``shape`` - a 2-tuple
         where ``None`` stands for unspecified, ie. ``(None, 2)`` will match
         any array where the second dimension is 2.
       - If ``m`` is a 1D array its shape is checked to see if it can be
         reshaped to ``shape``.  A n-array could be reshaped as (n,1) or (1,n)
         or any other shape with the correct number of elements.  A value of
         ``None`` in the shape stands for unspecified, ie. ``(None, 2)`` will 
         attempt to reshape ``m`` as an array with shape (k,2) where :math:`k \times 2 \eq n`.
       - If ``m`` is a scalar, return an array of shape (1,1)

    :seealso: :func:`ismatrix`, :func:`verifymatrix`
    """
    if isinstance(m, np.ndarray) and len(m.shape) == 2:
        # passed a 2D array
        mshape = m.shape

        if (shape[0] is None or shape[0] == mshape[0]) and (shape[1] is None or shape[1] == mshape[1]):
                return m
        else:
            raise ValueError(f"expecting {shape} but got {mshape}")

    elif isvector(m):
            # passed a 1D array
            m = getvector(m)
            if shape[0] is not None and shape[1] is not None:
                if len(m) == np.prod(shape):
                    return m.reshape(shape)
                else:
                    raise ValueError('array cannot be reshaped')
            elif shape[0] is not None and shape[1] is None:
                return m.reshape((shape[0], -1))
            elif shape[0] is None and shape[1] is not None:
                return m.reshape((-1, shape[1]))
            else:
                return m.reshape((1, -1))

    else:
        raise TypeError('argument must be scalar or ndarray')`

It seems the problem should be the input value type. Could you please give any advice to solve it?

YijiongLin, 2020-12-14:
[It disappeared when using release version v0.6.1.](https://github.com/petercorke/robotics-toolbox-python/issues/159)

closed 2020-12-14 by YijiongLin

petercorke, 2020-12-14:
Thanks for your interest and the feedback, we're still developing and occasionally things get temporarily broken.  Please keep in touch.