petercorke/robotics-toolbox-python #164: Error when calling newest DH models

groslombaire, 2021-01-11:
Hi,

First of all thank you for all the work you put in this toolbox !

I tried to use the new DH models (Hyper, Coil, Sawyer, Mico, Jaco, UR3, UR5, UR10) using this code :
```
import roboticstoolbox as rtb

robot = rtb.models.DH.Coil()
print(robot)
```

But it doesn't work and I get a message like this for all of these models :
`AttributeError: module 'roboticstoolbox.models.DH' has no attribute 'Coil'`

I ran into this error as I tried to make my own DH model. I don't know if if i am doing it wrong as I am a beginner.

Thank you for your time

groslombaire, 2021-01-12:
I just figured out that I was using an outdated version of the toolbox installed via Pycharm, problem solved !

closed 2021-01-12 by groslombaire

petercorke, 2021-01-15:
Thanks for your interest.  The rate of change is slowing,but feel free to send questions or bug reports