petercorke/robotics-toolbox-python #166: ikine_unc() function not found

vpadois, 2021-01-21:
Thank for this conversion of the RTB to Python. When running the example provided in the Readme, I get the following message:

 `AttributeError: 'Panda' object has no attribute 'ikine_unc'`

The remainer of the example code runs perfectly smoothly.

Thanks in advance for your help.

randomcascade, 2021-01-22:
Looks as if ikine_unc and ikine_con are being folded into a new method called ikine_min.
[https://github.com/petercorke/robotics-toolbox-python/commit/e8a9cf17137dff73c44abfd76d4b6acddd907bba](url)
Looks like running ikine_min with the "SLSQP" as the method attribute should run similarly if I'm not mistaken (and I may be).

vpadois, 2021-01-22:
Thanks for your feedback. Looking deeper in the available methods I found ikunc which seems to be the function I was looking for. Next time I will dig a bit more.

Cheers

closed 2021-01-22 by vpadois

petercorke, 2021-01-22:
Correct about ik_min.  The _con and _unc variants came across from the MATLAB version but they differed by just a line or two so I merged them.  I made some notes and timing info on the wiki https://github.com/petercorke/robotics-toolbox-python/wiki/Kinematics

I will update the examples.