petercorke/robotics-toolbox-python #169: Building wheel for roboticstoolbox-python (setup.py) ... error

Kanglatae, 2021-01-31:
Microsoft Windows [Version 10.0.19042.746]
(c) 2020 Microsoft Corporation. All rights reserved.

C:\Users\Konard>pip3 install roboticstoolbox-python
Collecting roboticstoolbox-python
  Using cached roboticstoolbox-python-0.7.0.tar.gz (234 kB)
Requirement already satisfied: numpy>=1.18.0 in c:\users\konard\appdata\local\programs\python\python38-32\lib\site-packages (from roboticstoolbox-python) (1.19.2)
Requirement already satisfied: spatialmath-python>=0.8.7 in c:\users\konard\appdata\local\programs\python\python38-32\lib\site-packages (from roboticstoolbox-python) (0.8.8)
Requirement already satisfied: pgraph-python in c:\users\konard\appdata\local\programs\python\python38-32\lib\site-packages (from roboticstoolbox-python) (0.6.1)
Requirement already satisfied: scipy in c:\users\konard\appdata\local\programs\python\python38-32\lib\site-packages (from roboticstoolbox-python) (1.5.2)
Requirement already satisfied: matplotlib in c:\users\konard\appdata\local\programs\python\python38-32\lib\site-packages (from roboticstoolbox-python) (3.3.2)
Requirement already satisfied: ansitable in c:\users\konard\appdata\local\programs\python\python38-32\lib\site-packages (from roboticstoolbox-python) (0.9.5)
Requirement already satisfied: swift-sim>=0.6.1 in c:\users\konard\appdata\local\programs\python\python38-32\lib\site-packages (from roboticstoolbox-python) (0.7.0)
Requirement already satisfied: qpsolvers in c:\users\konard\appdata\local\programs\python\python38-32\lib\site-packages (from roboticstoolbox-python) (1.5)
Requirement already satisfied: colored in c:\users\konard\appdata\local\programs\python\python38-32\lib\site-packages (from spatialmath-python>=0.8.7->roboticstoolbox-python) (1.4.2)
Requirement already satisfied: websockets in c:\users\konard\appdata\local\programs\python\python38-32\lib\site-packages (from swift-sim>=0.6.1->roboticstoolbox-python) (8.1)
Requirement already satisfied: cycler>=0.10 in c:\users\konard\appdata\local\programs\python\python38-32\lib\site-packages (from matplotlib->roboticstoolbox-python) (0.10.0)
Requirement already satisfied: kiwisolver>=1.0.1 in c:\users\konard\appdata\local\programs\python\python38-32\lib\site-packages (from matplotlib->roboticstoolbox-python) (1.2.0)
Requirement already satisfied: pyparsing!=2.0.4,!=2.1.2,!=2.1.6,>=2.0.3 in c:\users\konard\appdata\local\programs\python\python38-32\lib\site-packages (from matplotlib->roboticstoolbox-python) (2.4.7)
Requirement already satisfied: python-dateutil>=2.1 in c:\users\konard\appdata\local\programs\python\python38-32\lib\site-packages (from matplotlib->roboticstoolbox-python) (2.8.1)
Requirement already satisfied: pillow>=6.2.0 in c:\users\konard\appdata\local\programs\python\python38-32\lib\site-packages (from matplotlib->roboticstoolbox-python) (7.2.0)
Requirement already satisfied: certifi>=2020.06.20 in c:\users\konard\appdata\local\programs\python\python38-32\lib\site-packages (from matplotlib->roboticstoolbox-python) (2020.6.20)
Requirement already satisfied: six in c:\users\konard\appdata\local\programs\python\python38-32\lib\site-packages (from cycler>=0.10->matplotlib->roboticstoolbox-python) (1.15.0)
Requirement already satisfied: quadprog>=0.1.8 in c:\users\konard\appdata\local\programs\python\python38-32\lib\site-packages (from qpsolvers->roboticstoolbox-python) (0.1.8)
Requirement already satisfied: Cython in c:\users\konard\appdata\local\programs\python\python38-32\lib\site-packages (from quadprog>=0.1.8->qpsolvers->roboticstoolbox-python) (0.29.21)
Building wheels for collected packages: roboticstoolbox-python
  Building wheel for roboticstoolbox-python (setup.py) ... error
  ERROR: Command errored out with exit status 1:
   command: 'c:\users\konard\appdata\local\programs\python\python38-32\python.exe' -u -c 'import sys, setuptools, tokenize; sys.argv[0] = '"'"'C:\\Users\\Konard\\AppData\\Local\\Temp\\pip-install-a5_4sl4u\\roboticstoolbox-python_a8b4fafd0abd4f878182ccc0f67274b3\\setup.py'"'"'; __file__='"'"'C:\\Users\\Konard\\AppData\\Local\\Temp\\pip-install-a5_4sl4u\\roboticstoolbox-python_a8b4fafd0abd4f878182ccc0f67274b3\\setup.py'"'"';f=getattr(tokenize, '"'"'open'"'"', open)(__file__);code=f.read().replace('"'"'\r\n'"'"', '"'"'\n'"'"');f.close();exec(compile(code, __file__, '"'"'exec'"'"'))' bdist_wheel -d 'C:\Users\Konard\AppData\Local\Temp\pip-wheel-bfnvoy1j'
       cwd: C:\Users\Konard\AppData\Local\Temp\pip-install-a5_4sl4u\roboticstoolbox-python_a8b4fafd0abd4f878182ccc0f67274b3\
  Complete output (154 lines):
  running bdist_wheel
  running build
  running build_py
  creating build
  creating build\lib.win32-3.8
  creating build\lib.win32-3.8\roboticstoolbox
  copying roboticstoolbox\__init__.py -> build\lib.win32-3.8\roboticstoolbox
  creating build\lib.win32-3.8\roboticstoolbox\backends
  copying roboticstoolbox\backends\Connector.py -> build\lib.win32-3.8\roboticstoolbox\backends
  copying roboticstoolbox\backends\__init__.py -> build\lib.win32-3.8\roboticstoolbox\backends
  creating build\lib.win32-3.8\roboticstoolbox\mobile
  copying roboticstoolbox\mobile\animations.py -> build\lib.win32-3.8\roboticstoolbox\mobile
  copying roboticstoolbox\mobile\bug2.py -> build\lib.win32-3.8\roboticstoolbox\mobile
  copying roboticstoolbox\mobile\drivers.py -> build\lib.win32-3.8\roboticstoolbox\mobile
  copying roboticstoolbox\mobile\dstar.py -> build\lib.win32-3.8\roboticstoolbox\mobile
  copying roboticstoolbox\mobile\dx_form.py -> build\lib.win32-3.8\roboticstoolbox\mobile
  copying roboticstoolbox\mobile\landmarkmap.py -> build\lib.win32-3.8\roboticstoolbox\mobile
  copying roboticstoolbox\mobile\navigation.py -> build\lib.win32-3.8\roboticstoolbox\mobile
  copying roboticstoolbox\mobile\PoseGraph.py -> build\lib.win32-3.8\roboticstoolbox\mobile
  copying roboticstoolbox\mobile\sensors.py -> build\lib.win32-3.8\roboticstoolbox\mobile
  copying roboticstoolbox\mobile\vehicle.py -> build\lib.win32-3.8\roboticstoolbox\mobile
  copying roboticstoolbox\mobile\__init__.py -> build\lib.win32-3.8\roboticstoolbox\mobile
  creating build\lib.win32-3.8\roboticstoolbox\models
  copying roboticstoolbox\models\list.py -> build\lib.win32-3.8\roboticstoolbox\models
  copying roboticstoolbox\models\__init__.py -> build\lib.win32-3.8\roboticstoolbox\models
  creating build\lib.win32-3.8\roboticstoolbox\robot
  copying roboticstoolbox\robot\DHFactor.py -> build\lib.win32-3.8\roboticstoolbox\robot
  copying roboticstoolbox\robot\DHLink.py -> build\lib.win32-3.8\roboticstoolbox\robot
  copying roboticstoolbox\robot\DHRobot.py -> build\lib.win32-3.8\roboticstoolbox\robot
  copying roboticstoolbox\robot\Dynamics.py -> build\lib.win32-3.8\roboticstoolbox\robot
  copying roboticstoolbox\robot\ELink.py -> build\lib.win32-3.8\roboticstoolbox\robot
  copying roboticstoolbox\robot\ERobot.py -> build\lib.win32-3.8\roboticstoolbox\robot
  copying roboticstoolbox\robot\ETS.py -> build\lib.win32-3.8\roboticstoolbox\robot
  copying roboticstoolbox\robot\ETSRobot.py -> build\lib.win32-3.8\roboticstoolbox\robot
  copying roboticstoolbox\robot\Gripper.py -> build\lib.win32-3.8\roboticstoolbox\robot
  copying roboticstoolbox\robot\IK.py -> build\lib.win32-3.8\roboticstoolbox\robot
  copying roboticstoolbox\robot\ikine_evaluate.py -> build\lib.win32-3.8\roboticstoolbox\robot
  copying roboticstoolbox\robot\ikine_evaluate2.py -> build\lib.win32-3.8\roboticstoolbox\robot
  copying roboticstoolbox\robot\Link.py -> build\lib.win32-3.8\roboticstoolbox\robot
[... 256 lines cut ...]
    copying roboticstoolbox\tools\xacro\xmlutils.py -> build\lib.win32-3.8\roboticstoolbox\tools\xacro
    copying roboticstoolbox\tools\xacro\__init__.py -> build\lib.win32-3.8\roboticstoolbox\tools\xacro
    running egg_info
    writing roboticstoolbox_python.egg-info\PKG-INFO
    writing dependency_links to roboticstoolbox_python.egg-info\dependency_links.txt
    writing requirements to roboticstoolbox_python.egg-info\requires.txt
    writing top-level names to roboticstoolbox_python.egg-info\top_level.txt
    reading manifest file 'roboticstoolbox_python.egg-info\SOURCES.txt'
    writing manifest file 'roboticstoolbox_python.egg-info\SOURCES.txt'
    creating build\lib.win32-3.8\roboticstoolbox\core
    copying roboticstoolbox\core\frne.c -> build\lib.win32-3.8\roboticstoolbox\core
    copying roboticstoolbox\core\ne.c -> build\lib.win32-3.8\roboticstoolbox\core
    copying roboticstoolbox\core\vmath.c -> build\lib.win32-3.8\roboticstoolbox\core
    running build_ext
[... cut ...] hon_a8b4fafd0abd4f878182ccc0f67274b3\\setup.py'"'"';f=getattr(tokenize, '"'"'open'"'"', open)(__file__);code=f.read().replace('"'"'\r\n'"'"', '"'"'\n'"'"');f.close();exec(compile(code, __file__, '"'"'exec'"'"'))' install --record 'C:\Users\Konard\AppData\Local\Temp\pip-record-z9mpqhty\install-record.txt' --single-version-externally-managed --compile --install-headers 'c:\users\konard\appdata\local\programs\python\python38-32\Include\roboticstoolbox-python' Check the logs for full command output.

Jakubach, 2021-02-01:
I had the same error and I fixed it by installing from source with command:
`pip3 install git+https://github.com/petercorke/robotics-toolbox-python`

Kanglatae, 2021-02-01:
> I had the same error and I fixed it by installing from source with command:
> `pip3 install git+https://github.com/petercorke/robotics-toolbox-python`

Worked. Thank you!

petercorke, 2021-02-03:
What platform?

Kanglatae, 2021-02-03:
> What platform?

Windows.

petercorke, 2021-02-07:
Fixed now.  For some reason your install was building locally, even though we ship a wheel for Windows.  There was a missing .h file in the package which is why you got the error.

closed 2021-02-07 by petercorke

jhavl, 2021-02-08:
This was fixed with v0.8.0 which has been pushed to PyPI. Extra files were not being included in the source dist.

MCollins52, 2021-04-13:
I'm running into the same (or very similar) issue. I tried installing from source which is mentioned above but no luck. I'm on Windows 10 with Python 3.9. Any help?

jhavl, 2021-04-13:
Can you please attach the error output?

MCollins52, 2021-04-13:
> Can you please attach the error output?

[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]

jhavl, 2021-04-14:
Thanks for that. This is a different issue to the one above. It seems that the build tools can't locate the standard library math.h. This issue should be fixable by re-installing the C++ build tools which can be downloaded [here](https://visualstudio.microsoft.com/visual-cpp-build-tools/).

Otherwise, the PyPI version of the toolbox is pre-build so not build tools are required. We will be pushing a new version of the toolbox to PypI this week so it will be up to date.

MCollins52, 2021-04-14:
> Thanks for that. This is a different issue to the one above. It seems that the build tools can't locate the standard library math.h. This issue should be fixable by re-installing the C++ build tools [...]

I installed the build tools and that didn't seem to work, so I updated the C++ build tools that I currently had installed and installed the toolbox using pip, and that seemed to work. Thanks!