petercorke/robotics-toolbox-python #170: Swift not rendering correctly

Jakubach, 2021-02-03:
Hello, 
I am using Robotic Toolbox in Pycharm on Windows 10. When I run this example code:
```
import roboticstoolbox as rtb
import spatialmath as sm
import numpy as np

# Make and instance of the Swift simulator and open it
env = rtb.backends.Swift()
env.launch()

# Make a panda model and set its joint angles to the ready joint configuration
panda = rtb.models.Panda()
panda.q = panda.qr

# Set a desired and effector pose an an offset from the current end-effector pose
Tep = panda.fkine() * sm.SE3.Tx(0.2) * sm.SE3.Ty(0.2) * sm.SE3.Tz(0.45)

# Add the robot to the simulator
env.add(panda)

# Simulate the robot while it has not arrived at the goal
arrived = False
while not arrived:
    # Work out the required end-effector velocity to go towards the goal
    v, arrived = rtb.p_servo(panda.fkine(), Tep, 1)

    # Set the Panda's joint velocities
    panda.qd = np.linalg.pinv(panda.jacobe()) @ v

    # Step the simulator by 50 milliseconds
    env.step(0.05)
```
then my Firefox browser opens a new cart and I can see only the coordinate frame, simulations flows and then exit. The robot isn't shown. I tried also with UR10 model and had the same issue.

Jakubach, 2021-02-03:
Fixed by correcting an error in example code presented in [swift markdown](https://github.com/jhavl/swift)
Corrected by adding joint configuration vector as argument for fkine() function.
```
import roboticstoolbox as rtb
import spatialmath as sm
import numpy as np

# Make and instance of the Swift simulator and open it
env = rtb.backends.Swift()
env.launch()

# Make a panda model and set its joint angles to the ready joint configuration
panda = rtb.models.Panda()
panda.q = panda.qr

# Set a desired and effector pose an an offset from the current end-effector pose
Tep = panda.fkine(panda.q) * sm.SE3.Tx(0.2) * sm.SE3.Ty(0.2) * sm.SE3.Tz(0.45)

# Add the robot to the simulator
env.add(panda)

# Simulate the robot while it has not arrived at the goal
arrived = False
while not arrived:
    # Work out the required end-effector velocity to go towards the goal
    v, arrived = rtb.p_servo(panda.fkine(panda.q), Tep, 1)

    # Set the Panda's joint velocities
    panda.qd = np.linalg.pinv(panda.jacobe(panda.q)) @ v

    # Step the simulator by 50 milliseconds
    env.step(0.05)
```

closed 2021-02-03 by Jakubach

jhavl, 2021-02-08:
@Jakubach thanks for bringing this to our attention, I've fixed up the example on the Swift readme