petercorke/robotics-toolbox-python #171: Elbow-up/elbow-down configuration issue

Jakubach, 2021-02-04:
Hello,
I am testing on Windows 10 the Puma560 robot.
Plotting puma's qn position I get elbow-up configuration:
```
import roboticstoolbox as rtb
robot = rtb.models.DH.Puma560()
robot.plot(robot.qn)
```
[image]

Then, I solve forward kinematics to get transformation matrix.
`T = robot.fkine(robot.qn)`
And solve inverse kinematics for the same position and orientation but with another configuration (elbow-down):
`qd = robot.ikine_a(T = T, config="rd")`
But resulting configuration is still elbow-up.
qd: [-2.77555756e-17  7.85398163e-01  3.14159265e+00  3.14159265e+00
 -7.85398163e-01 -3.14159265e+00]
Changing to `qd = robot.ikine_a(T = T, config="ru")` which should result in elbow-up configuration gives me elbow-down config.
qd:[-2.77555756e-17 -8.33533063e-01  9.39558327e-02  1.05479583e-17
 -8.31219097e-01 -5.23446762e-17]

petercorke, 2021-02-05:
Thanks for letting me know.   The MATLAB code does this as well, which means the port is accurate :)  I know the MATLAB code used to get this right, so it'll take me a while to get to the bottom of.  This is amongst some of the oldest code in RTB...

petercorke, 2021-02-05:
Fixed on GH.

closed 2021-02-05 by petercorke