petercorke/robotics-toolbox-python #173: How could one plot rtb.models.DH.plot and spatialmath.base.trplot() in the same figure?

rojas70, 2021-02-10:
For the inverse kinematics portion, in matlab it was easy to plot both the robot's current configuration as well as the desired pose configuration. 

I have not found a way to fix the figure upon which both of these methods plot...

petercorke, 2021-02-11:
Pls send a minimal example that shows the problem.

rojas70, 2021-02-22:
````
# Imports
import math
import numpy as np

from spatialmath import *
from spatialmath.base import *

import roboticstoolbox as rtb
from roboticstoolbox import *

import matplotlib.pyplot as plt

# Puma
puma = rtb.models.DH.Puma560()

# Start at qn and plot 
puma.plot(puma.qn)

# Desired pose and plot
T = SE3(0.6, 0.1, 0.5)*SE3().RPY([0, 0, 0], unit='deg')
trplot(T.A, dims=[-2,2], frame='EE', rviz=True)                 # plots not superposed here.

# Desired angles... analytic/numerical solns
sol = puma.ikine_LM(T, q0=puma.qn)
puma.plot(sol.q);                                                              # Would like to see the robot des pose to be superposed over the trplot frame version. 
````

petercorke, 2021-02-23:
I get this

The axes are a bit big, but otherwise OK.  

Maybe do

puma.plot(puma.qn, block=False)

So that you don’t have to dismiss the Puma plot before doing the next plot.

At the moment, by default all RTB plots are blocking, but I’m not sure now that’s such a great idea.  Let me know how it goes.

Peter

rojas70, 2021-02-23:
Thank you. 

The block=False, works nicely for the first part, however, the second puma.plot resets the figure and overwrites it... It would be nice to use the same figure/axes there.