petercorke/robotics-toolbox-python #177: It is not possible to add new URDF robots in different path of models

olmerg, 2021-02-17:
I am giving my robotic class using this amazing tool in python library, but I found a problem. In an exercise, I want that student use a specific *urdf*, but It is not possible to view the mesh because Mesh.filename use an absolute path `àbspath`  to the models of the toolbox.
I know that this is the only error, because I change to the specific path and works, but I do not find which could be the best solution in the architecture of your code

https://github.com/petercorke/robotics-toolbox-python/blob/276efaaf3c67558cf2ef17e0360b6821fb8d7bfd/roboticstoolbox/tools/urdf/urdf.py#L301

petercorke, 2021-02-17:
Good point.  How would you like to be able to write the code you share with your students.  Give an example and we'll see what we can do! It's an important use case.

olmerg, 2021-02-17:
[In this link](https://drive.google.com/file/d/1mhyTkbRNooFU9Dp547Avf_WgOIPiLHSw/view?usp=sharing) you can download the minimal example.
The robot is a planar robot, which I take from an old tutorial of [ros industrial](https://github.com/ros-industrial/industrial_training) and in  [lesson_urdf](https://github.com/olmerg/lesson_urdf) I am trying to port to ros2.

mfkenson, 2021-02-19:
Good news is I could load your mesh files after few tricks using the given minimal example.
1. Specifiying the absolute path 
https://github.com/petercorke/robotics-toolbox-python/blob/276efaaf3c67558cf2ef17e0360b6821fb8d7bfd/roboticstoolbox/tools/urdf/urdf.py#L307
using `value = os.path.abspath(value)`

2. Move the "./xacro/lesson_urdf" folder to "./lesson_urdf"

This is a temporary workaround but shows a way to improve the toolbox later.
e.g. if external file not exist->fallback to looking at predefined base_path from the toolbox.

[image]

petercorke, 2021-02-19:
Ok, so default to current folder, then fallback to the distribution.  We are thinking about putting the models in a separate package, they make the RTB package very large and don't change much.  This won't affect operation from user perspective.

mcevoyandy, 2021-04-07:
I'm also having issues loading my own URDF. 

I created my own class using the other URDF python files as an example and then added this to the __init__.py in the URDF folder. No errors appear when loading the robot but when I try to run `backend.add(robot)` using Swift() nothing displays. 

I also just pulled master and see you've removed Swift...

At any rate, I'm very interested to see how to load a custom URDF and get it to display.

olmerg, 2021-04-29:
I continue having the problem with my own urdf when I try to use a personalized swift backend.
I have test my urdf with swift and works ok [in this file](https://github.com/olmerg/rtb_serial_robot/blob/main/planar_3dof/example.py) but then I create a new backend which inherits from swift to sent by pyserial data to an arduino. The new backend is [this file](https://github.com/olmerg/rtb_serial_robot/blob/main/Swift_serial/Swift_serial.py).

If I test my backend with a library robot([UR3 test](https://github.com/olmerg/rtb_serial_robot/blob/main/example.py)) and my arduino robot and works, by the idea is that the student has created his own urdf  (real robot too). 

The problem is when I try to work with my custom urdf with my custom swift backend ([example](https://github.com/olmerg/rtb_serial_robot/blob/main/planar_3dof/example_swift_serial.py)) it does not find the urdf file and said when execute add:
```
  File "C:\Users\olmer\OneDrive - Ocox\Documents\2021\robIndustrial\Swift_serial\planar_3dof\Planar3DOF.py", line 13, in __init__
    args = super().urdf_to_ets_args(
  File "C:\Users\olmer\anaconda3\lib\site-packages\roboticstoolbox\robot\ERobot.py", line 410, in urdf_to_ets_args
    urdf = URDF.loadstr(open(file_path).read(), file_path)
FileNotFoundError: [Errno 2] No such file or directory: 'C:\\Users\\olmer\\anaconda3\\Lib\\site-packages\\rtbdata\\xacro\\lesson_urdf\\urdf\\planar_3dof.urdf'
```
What I understand is that it is looking the folder xacro but in this moment is executing the swift_serial backend so he try to open the library path.

olmerg, 2021-04-29:
I found the problem, I made launch before add.

@mcevoyandy  May be is your problem too, first use env.add and then add.launch

petercorke, 2021-04-30:
Launch() should be called before add().  Are you saying that ordering causes an error?

peter

olmerg, 2021-04-30:
yes, this was the solution, maybe is my implementation, because launch call reset, which in my implementation send to home my robot and reset the robot position.

olmerg, 2021-05-20:
I reopen the issue because the solution implemented works if I execute a python, but with a notebook, it opens the first moment, but if I try to create again the robot object(or create another robot) it does not detect the folder xacro. The problem is that the path of ipython is not the same of the ipyn file. 
maybe this can help[ipython](https://github.com/ipython/ipython/issues/10123)

olmerg, 2021-05-20:
.

jhavl, 2021-05-31:
Hey @olmerg, I'm unable to reproduce the error, I have started a Jupyter Notebook in my home directory (outside of the Toolbox Python directory) and include the code:
```
import roboticstoolbox as rtb
import swift

r = rtb.models.Panda()
env = swift.Swift()
env.launch()
env.add(r)
```
and it works correctly. Are you on the latest version of the toolbox?

olmerg, 2021-05-31:
This are the conditions:

- you have a xacro folder with a custom urdf project in the place of the ipynb 
- you create a robot in a cell (the same that you made)
- first time that you execute the notebook it works, but **if you execute again the cell , the xacro folder is not found and it try to go to default folder**.

- the temporal solution is restart the kernel

The problem is that the path of ipython is not the same of the ipynb file.

petercorke, 2021-05-31:
@jhavl do you do any cd command perhaps. As part of swift startup?

@olmerg what OS?

mfkenson, 2021-05-31:
I can reproduce the issue. Not only in jupyter notebook.
Python3.6; Ubuntu 18.04. latest commit of robotics-toolbox-python (338d960) and swift (2a0c93e).

urdf from @olmerg previous comment

```
import swift
import numpy as np
from roboticstoolbox.robot.ERobot import ERobot
from math import pi

class Planar3DOF(ERobot):

    def __init__(self):
        links, name = self.URDF_read(
            "lesson_urdf/urdf/planar_3dof.urdf"
        )

        super().__init__(
            links,
            name=name.upper(),
            manufacturer='Custom Unamed'
        )

        self.addconfiguration("qz", np.array([0, 0, 0]))
        self.addconfiguration("qr", np.array([0, pi / 2, -pi / 2]))
        self.addconfiguration("qs", np.array([0, 0, -pi / 2]))
        self.addconfiguration("qn", np.array([0, pi / 4, pi]))

r1 = Planar3DOF() # ok here
env = swift.Swift()
env.launch()
r2 = Planar3DOF() # failed here
```

jhavl, 2021-05-31:
Thanks for the details @olmerg, @mfkenson, I will investigate further.

mfkenson, 2021-05-31:
I digged into the debugger and believe the difference between r1 and r2 is the result of resolving "path_to_datafile" and this line is suspicious. https://github.com/petercorke/robotics-toolbox-python/blob/71183f3221cf9ced69420f07ade06f4514c256ba/roboticstoolbox/tools/data.py#L88

1st time it can return the user path
[image]

2nd time it failed
[image]

mfkenson, 2021-05-31:
@olmerg  workaround / temporary fix for now:

```
import swift
import numpy as np
from roboticstoolbox.robot.ERobot import ERobot
from math import pi

class Planar3DOF(ERobot):

    def __init__(self):
        links, name = self.URDF_read(
            "lesson_urdf/urdf/planar_3dof.urdf"
        )

        super().__init__(
            links,
            name=name.upper(),
            manufacturer='Custom Unamed'
        )

        self.addconfiguration("qz", np.array([0, 0, 0]))
        self.addconfiguration("qr", np.array([0, pi / 2, -pi / 2]))
        self.addconfiguration("qs", np.array([0, 0, -pi / 2]))
        self.addconfiguration("qn", np.array([0, pi / 4, pi]))

r1 = Planar3DOF()

cwd = os.getcwd()

env = swift.Swift()
env.launch()

os.chdir(cwd)

r2 = Planar3DOF() # ok again
env.add(r2)
```

kxxt, 2021-06-06:
Well, I think it's because `env.launch()` changed the current working dir to the root of the disk. 
```python
>>> from os import getcwd
>>> getcwd()
'C:\\Users\\XXXX'
>>> backend = Swift()
>>> getcwd()
'C:\\Users\\XXXX'
>>> backend.launch()
>>> getcwd()
'C:\\'
```

https://github.com/jhavl/swift/blob/2a0c93e604089f0c8086a03d19521b4d15169a8f/swift/SwiftRoute.py#L154

kxxt, 2021-06-06:
@mfkenson  However, changing the current working directory back would result in a 404 error page if the swift page get refreshed.

kxxt, 2021-06-06:
I've worked out a solution. No more need for `chdir`. 
My modified version: https://github.com/kxxt/swift/
I'll open a pull request soon.

jhavl, 2021-06-07:
This should be fixed on the `future` branch of swift thanks to a pull request from @kxxt. Let me know if you still have any issues @olmerg

closed 2021-06-07 by jhavl