petercorke/robotics-toolbox-python #179: Adding mesh shapes to environment

eastSpawn, 2021-02-18:
I am trying to add STL mesh shapes to the swift environment without any success. Here is the code I am trying to use (taken from example gimbal.py):

```
fig = rtb.Mesh(
    filename=str('fig.stl'),
    base=sm.SE3(0, 0, 1.3),
    color=[34, 143, 201]
)
env.add(fig)
env.step()
```
Has this been properly implemented yet? or is there something wrong with the approach?

petercorke, 2021-02-19:
This is new capability but your example should work, it does in gimbal.  What's your issue?

eastSpawn, 2021-02-19:
My bad! The example works for me now. The path to the STL files didn't work properly in my setup when running it directly from the example. Thank you

closed 2021-02-19 by eastSpawn