petercorke/robotics-toolbox-python #182: Robot in Swift doesn't move

BartlomiejKulecki, 2021-02-25:
Hello,
I am using Robotic Toolbox in Pycharm on Windows 10. When I run this example code:
```
import roboticstoolbox as rtb
import spatialmath as sm
import numpy as np

# Make and instance of the Swift simulator and open it
env = rtb.backends.Swift()
env.launch()

# Make a panda model and set its joint angles to the ready joint configuration
panda = rtb.models.Panda()
panda.q = panda.qr

# Set a desired and effector pose an an offset from the current end-effector pose
Tep = panda.fkine(panda.q) * sm.SE3.Tx(0.2) * sm.SE3.Ty(0.2) * sm.SE3.Tz(0.45)

# Add the robot to the simulator
env.add(panda)

# Simulate the robot while it has not arrived at the goal
arrived = False
while not arrived:
    # Work out the required end-effector velocity to go towards the goal
    v, arrived = rtb.p_servo(panda.fkine(panda.q), Tep, 1)

    # Set the Panda's joint velocities
    panda.qd = np.linalg.pinv(panda.jacobe(panda.q)) @ v

    # Step the simulator by 50 milliseconds
    env.step(0.05)
```
the browser window opens, the robot renders correctly but it doesn't move.

jhavl, 2021-03-01:
Which version of the toolbox and Swift are you ruunning?
Are you installed off PyPI or from Github?
Whats your Python version?
Any error messages in your terminal or browser console? (Open browser console with control+shift+i)

BartlomiejKulecki, 2021-03-01:
I just reinstalled toolbox with the commands:
```
pip3 uninstall roboticstoolbox-python 
pip3 install git+https://github.com/petercorke/robotics-toolbox-python
```
Python version 3.7
In terminal no errors.
In browser console there is one error:

[image]

mfkenson, 2021-03-02:
@jhavl I could reproduce the issue with latest commit 448247e in master branch.
I tried to identify which section causing the issue and found 
`self.process_events()`==>`events = self._send_socket('check_elements')` ==> `return self.inq.get()` is blocking.

If I comment out this line 158. The robot moves as expected.
https://github.com/petercorke/robotics-toolbox-python/blob/448247e40e0b1225b77de70b8a6cf1f299c8a5b9/roboticstoolbox/backends/Swift/Swift.py#L158
https://github.com/petercorke/robotics-toolbox-python/blob/448247e40e0b1225b77de70b8a6cf1f299c8a5b9/roboticstoolbox/backends/Swift/Swift.py#L502

It seems the queue.get is blocking by default. Let me try to pass `block=False` and catch the empty exception.

jhavl, 2021-03-03:
@mfkenson thanks for this, browser cache is a common issue I run into. Setting no-cache is probably the best solution at least while Swift is under rapid development. I also added some basic troubleshooting tips to the [common issues wiki page](https://github.com/petercorke/robotics-toolbox-python/wiki/Common-Issues).

closed 2021-03-03 by jhavl