petercorke/robotics-toolbox-python #184: Teach function crashes

BartlomiejKulecki, 2021-02-25:
Hello
I have a problem with function teach(). After few moves of slider there are errors and window closes.
Error looks like this:
```
Exception in Tkinter callback
Traceback (most recent call last):
  File "C:\Users\Y700\AppData\Local\Programs\Python\Python37\lib\tkinter\__init__.py", line 1705, in __call__
    return self.func(*args)
  File "C:\Users\Y700\AppData\Local\Programs\Python\Python37\lib\tkinter\__init__.py", line 749, in callit
    func(*args)
  File "C:\Users\Y700\AppData\Local\Programs\Python\Python37\lib\site-packages\matplotlib\backends\_backend_tk.py", line 338, in idle_draw
    self.draw()
  File "C:\Users\Y700\AppData\Local\Programs\Python\Python37\lib\site-packages\matplotlib\backends\backend_tkagg.py", line 9, in draw
    super(FigureCanvasTkAgg, self).draw()
  File "C:\Users\Y700\AppData\Local\Programs\Python\Python37\lib\site-packages\matplotlib\backends\backend_agg.py", line 388, in draw
    self.figure.draw(self.renderer)
  File "C:\Users\Y700\AppData\Local\Programs\Python\Python37\lib\site-packages\matplotlib\artist.py", line 38, in draw_wrapper
    return draw(artist, renderer, *args, **kwargs)
  File "C:\Users\Y700\AppData\Local\Programs\Python\Python37\lib\site-packages\matplotlib\figure.py", line 1707, in draw
    self.patch.draw(renderer)
  File "C:\Users\Y700\AppData\Local\Programs\Python\Python37\lib\site-packages\matplotlib\artist.py", line 38, in draw_wrapper
    return draw(artist, renderer, *args, **kwargs)
  File "C:\Users\Y700\AppData\Local\Programs\Python\Python37\lib\site-packages\matplotlib\patches.py", line 571, in draw
    self._bind_draw_path_function(renderer) as draw_path:
  File "C:\Users\Y700\AppData\Local\Programs\Python\Python37\lib\contextlib.py", line 112, in __enter__
    return next(self.gen)
  File "C:\Users\Y700\AppData\Local\Programs\Python\Python37\lib\site-packages\matplotlib\patches.py", line 516, in _bind_draw_path_function
    gc = renderer.new_gc()
  File "C:\Users\Y700\AppData\Local\Programs\Python\Python37\lib\site-packages\matplotlib\backend_bases.py", line 656, in new_gc
    return GraphicsContextBase()
  File "C:\Users\Y700\AppData\Local\Programs\Python\Python37\lib\site-packages\matplotlib\backend_bases.py", line 731, in __init__
    self._hatch_color = colors.to_rgba(rcParams['hatch.color'])
  File "C:\Users\Y700\AppData\Local\Programs\Python\Python37\lib\site-packages\matplotlib\colors.py", line 167, in to_rgba
    if _is_nth_color(c):
  File "C:\Users\Y700\AppData\Local\Programs\Python\Python37\lib\site-packages\matplotlib\colors.py", line 117, in _is_nth_color
    return isinstance(c, str) and re.match(r"\AC[0-9]+\Z", c)
  File "C:\Users\Y700\AppData\Local\Programs\Python\Python37\lib\re.py", line 173, in match
    return _compile(pattern, flags).match(string)
RecursionError: maximum recursion depth exceeded
Fatal Python error: Cannot recover from stack overflow.
```
It occurs mainly after few times when I change values on slider with mouse button pressed. When I change value by clicking once on a slider, usually there are no problems.

petercorke, 2021-02-26:
What robot model and backend?

petercorke, 2021-02-26:
This seems familiar to me, from last year some time, maybe we regressed, or else it's a TkInter specific issue.

petercorke, 2021-02-26:
I've pushed a couple of changes to `teachbug` branch.  Basically `update()` no longer steps the environment, I think this is the issue.  `hold` now loops and drives the MPL backend event handler.  All good, but it's a mess when we close the window.  Need some way to detect that in the hold loop.

petercorke, 2021-02-26:
This might do the trick, but not for tonight

```
if plt.fignum_exists(<figure number>):
    # Figure is still opened
else:
    # Figure is closed
```

I'm hoping the figure number is stashed away in the environment somewhere, and we can check this in the hold loop.

petercorke, 2021-02-28:
Fixed on GH, will be in PyPI 0.8.1

closed 2021-02-28 by petercorke