petercorke/robotics-toolbox-python #185: VPython bug

BartlomiejKulecki, 2021-03-01:
Hello
When I run the code:
```python
import roboticstoolbox as rtb
from spatialmath import *

robot = rtb.models.DH.Panda()
T = SE3(0.7, 0.2, 0.1) * SE3.OA([0, 1, 0], [0, 0, -1])
sol = robot.ikine_LM(T)  # solve IK
q_pickup = sol.q
print(q_pickup)  # display joint angles
print(robot.fkine(q_pickup))
qt = jtraj(robot.qz, q_pickup, 50)

env = rtb.backends.VPython()
env.launch()
env.add(fig_num=0, dhrobot=robot)
for q in qt.y:
    time.sleep(1/50)
    env.step(dt=0.025, id=robot, q=q)
```
the browser opens, robot is rendering, but when it should start moving (it is not) there are errors:
```
File "D:/rtb/scripts/main.py", line 100, in traj_vpython
  env.step(dt=0.025, id=robot, q=q)
File "C:\Users\Y700\AppData\Local\Programs\Python\Python37\lib\site-packages\roboticstoolbox\backends\VPython\VPython.py", line 200, in step
  if robot is None:
UnboundLocalError: local variable 'robot' referenced before assignment
```

micah-huth, 2021-03-04:
This should be fixable by replacing \roboticstoolbox\backends\VPython\VPython.py line 190 from
"# robot = None" to
"robot = id".
This will allow following conditional statements to run correctly.

I can't test/implement this at the moment, but I can get to it (if no one else does) early next week.

petercorke, 2021-03-05:
@micah-huth good to hear from you, be great if you could have a look at this.  Few things is like to chat to you about if you have cycles...

PR #193 (2021-03-11): VPython: Robot variable reference fix
Closes #185
closed 2021-03-11 by micah-huth