petercorke/robotics-toolbox-python #186: robot.plot method never return in jupyter notebook

mfkenson, 2021-03-01:
```
%matplotlib notebook
import numpy as np
import math
from spatialmath import *
from spatialmath.base import *
from roboticstoolbox import *
from collections import namedtuple
import roboticstoolbox.tools.trajectory as tr
import matplotlib.pyplot as plt

np.set_printoptions(linewidth=100, formatter={'float': lambda x: f"{x:8.4g}" if abs(x) > 1e-10 else f"{0:8.4g}"})
from roboticstoolbox.models.DH import Panda

panda = Panda()
panda.plot(panda.q)
```
[image]

The while True here https://github.com/petercorke/robotics-toolbox-python/blob/master/roboticstoolbox/backends/PyPlot/PyPlot.py#L316.

mfkenson, 2021-03-01:
Submitted a PR for temporary quick fix. Hopefully there is a better way to handle it. Will look into this pyplot backend later in this week.

mfkenson, 2021-03-04:
workaround for now is simply passing block=False when calling the plot method.

e.g.
```
%matplotlib notebook
import matplotlib.pyplot as plt
from roboticstoolbox.models.DH import Panda

panda = Panda()
panda.plot(panda.q, block=False)
```

petercorke, 2021-03-05:
I am slowly changing the default for block to be false.  I do a lot of work in the python REPL so blocking is helpful, but otherwise is a nuisance.  There's a bit of changing going on with graphics at the minute.  Thanks for your interest and contributions.

mfkenson, 2021-03-05:
You are welcome. Thanks to @rojas70 that he introduced the toolbox to the robotics class here in CUHK. And I really enjoy to play around since then. My goal is to extend the toolbox to control a desktop low cost robot arm. Right now I am working on  firmware level to make it accept a trajectory feed through UART.

Instead of True/False, we might consider to make default=None and let the toolbox decide to block or not to block by default. Given that we could determine whether the runtime is in REPL or notebook.

petercorke, 2021-03-06:
Interesting idea, let me think about that. There is a plan for a physical robot backend, I started but got distracted, for Dynamixels. Be happy to chat more about that.

petercorke, 2021-03-06:
@rojas70 thanks for taking a gamble on a very new software tool.  Be happy to learn your perspectives about its use for teaching.

mfkenson, 2021-03-08:
I m working on a steppers based robot. My basic idea started by feeding the output of mstraj to arduino teensy. I can see the websocket approach from @jhavl swift. Somethings similar could be done over serial port.

There are few awesome libraries for arduino to precisely control multiple steppers ( position control with acceleration). Let’s build a bridge between them🚀🚀🚀

olmerg, 2021-03-13:
@mfkenson I am trying to make a backend to[ low cost servos robot](https://articulo.mercadolibre.com.co/MCO-523547836-brazo-robotico-arduino-7-servos-6-dof-grados-libertad-s3003-_JM#position=25&type=item&tracking_id=52d0dcf1-40d0-4c50-8ab0-870e39de35aa), which some student are building, I take the example swift backend and make something like  this in step() method of the backed:
```
                for i in range(robot.n):
                    q_1=self.q_1[i]
                    robot.q[i] += robot.qd[i] * (dt)

                    if np.any(robot.qlim[:, i] != 0) and \
                            not np.any(np.isnan(robot.qlim[:, i])):
                        robot.q[i] = np.min([robot.q[i], robot.qlim[1, i]])
                        robot.q[i] = np.max([robot.q[i], robot.qlim[0, i]])
                    move=int(round((robot.q[i]-q_1)*180.0/pi))
                    if(abs(move)>0):
                        self.move_serial(i,move)
                line=self.serial.readline().decode("utf-8")
                if line[0]=='*' and line[-3]=='*'
                    print(line[1:-3].split(','))
                #TODO: what to do if the motor did not come to the expected value?
                self.q_1=robot.q
```
The only component that I am using is pySerial, If you think that it can be useful I can share!

I have a big question about architecture, although these motors do not give feedback( we do not have sensors in Arduino to read the real position), how we can feedback some problem to robotics-toolbox-python, for example I send to move 10 degrees but he moves 8 degree by some problem o some not modeled, **we change q value of the toolbox robot or what we should do?**This question is open for me in the case I use other dynamics simulator or I use ROS (I think this could be other discussion).

mfkenson, 2021-03-14:
@olmerg IMHO have to admit that when dealing with systems without feedback, there is nothing can be done. Have to trust the system did what it did. I added a home/reset button so that I could manually let the robot go home and notify the python program to reset to home state. (In this case I am the feedback 😉 )

Yes changing the robot.q is a way to handle the feedback.

We could discuss more on this topic kenson@link.cuhk.edu.hk

olmerg, 2021-03-16:
thanks, @mfkenson  yes it is clear if you do not measure you can not have feedback, but I am simulating, because the robot also can be commanded by Bluetooth for a cell phone, so the idea is to teach some feedback concepts in class.
When I finish the protocol I will sent you , I believe in the feedback and may be can be useful:)

closed 2021-05-21 by petercorke