petercorke/robotics-toolbox-python #199: Calculate Forward and Inverse Kinematic from AR3

kstan79, 2021-03-15:
I'm trying to develop AR3 6DOF robot arm (https://www.anninrobotics.com/) web service from scratch with python, unfortunately I can't understand the existing forward/inverse kinematic source code cause too complex. I notice this toolbox can help calculate forward and inverse kinematic by provide DH parameter. 

Below is the DH parameter I get from their software:
a = [ 0.0642, 0.305, 0, 0, 0, 0 ]
d = [0.16977, 0, 0, -0.22263, 0, -0.03625 ]
alpha = [-pi / 2, zero, pi / 2, -pi / 2, pi / 2, zero]

I have 2 question hope there is solution for me
1. Can I calculate current xyz/rotation value by provide 6 joint angle? 
Example: [ 90, 0, 30, 45, 10, 0 ], desire answer something like [x:10, y: 0, z: 30....]

2. Can I calculate inverse kinematic like [x:10+5, y:0, z:40...], it can reverse calculate joint angle become something like: [100, 0, 40,50,60,0]

If above calculation is achievable, is there any sample we can refer cause I can't get from wiki

olmerg, 2021-03-17:
the idea create a class with the DH parameters, something like:
`class RobotScara(DHRobot):
    def __init__(self):
        super().__init__(
            [
                RevoluteDH(d=0.5,alpha=0,a=0),
                RevoluteDH(a=0.3,alpha=0,offset=0),
                PrismaticDH(a=0.2,alpha=-pi,offset=0.2),
                RevoluteDH(offset=0,d=0.0),
            ], name="Robot Scara")
        self.addconfiguration("qz", np.array([0, 0, 0, 0]))
        self.addconfiguration("qr", np.array([0, -pi/2, 0.5, 0]))
`

Where in addconfiguration you can some predertimate angles.

after that you can calculate the forward kinematics with 
`T = my_robot.fkine(np.array([0,0,0,0]))`
where the array is the value of the motors

for inverse kinematics you can review the wiki with functions like `ikne_lm`

kstan79, 2021-03-17:
hi olmerg,

Thanks reply, but sorry i really can't understand how the source code work, can elaborate more?
1. what is `qz` and what is the `[0,0,0,0]`?  
2. what is `qr` and what is `np.array([0, -pi/2, 0.5, 0]`?
And, where is the actual xyz location

Beside, i can't find any relevant content of `ikne_lm`, i search using github search tools only this keyword only exists in this issues, source code and wiki both not found.

kstan79, 2021-03-17:
ok, so `T = my_robot.fkine(np.array([0,0,0,0]))`, if I have 6 joint then i need to provide 6 joints value like `[0,90,70,180,0,0]`? It is degree or need to convert become radiant?

assume i'd get the `T`, I wish to move arm to x+3mm,Y+1mm,Z-2mm, since `T` is 4x4 matrix and how can I define new X,Y,Z?

olmerg, 2021-03-17:
1. yes you will have to give the six motor position (are in radian).
2. T is a matrix with orientation(3x3 matrix) and the position vector(px,py,pz) , but I recommend to review the packages spatialMath(also of perter corke) to understand better how to create this matrix.

kstan79, 2021-03-17:
Thanks reply, i will spend time to learn the basic of kinematic afterward. The position vector (px,py,pz) is the BLUE value in below image?

[image]

petercorke, 2021-03-18:
Yes, can extract position with T.t

kstan79, 2021-03-19:
I tried code in below method, is it correct?
```
import roboticstoolbox as rtb
from ar3model import AR3

robot = AR3()
print(robot)

rad = [0,0.1,0.2,0,0,0]
T = robot.fkine(rad)  # forward kinematics
print("Rad")
print(rad)
print("Initial T.t")
print(T.t)

T.t[0]+=0.010
T.t[1]+=0.001
T.t[0]-=0.002

print("New T.t")
print(T.t)
sol = robot.ikine_LM(T)
print(sol)
```

The result shown below:
```
┏━━━━┳━━━━━━━━━━┳━━━━━━━━┳━━━━━━━━┓
┃θⱼ  ┃    dⱼ    ┃   aⱼ   ┃   ⍺ⱼ   ┃
┣━━━━╋━━━━━━━━━━╋━━━━━━━━╋━━━━━━━━┫
┃ q1 ┃  0.16977 ┃ 0.0642 ┃ -90.0° ┃
┃ q2 ┃        0 ┃  0.305 ┃   0.0° ┃
┃ q3 ┃        0 ┃      0 ┃  90.0° ┃
┃ q4 ┃ -0.22263 ┃      0 ┃ -90.0° ┃
┃ q5 ┃        0 ┃      0 ┃  90.0° ┃
┃ q6 ┃ -0.03625 ┃      0 ┃   0.0° ┃
┗━━━━┻━━━━━━━━━━┻━━━━━━━━┻━━━━━━━━┛

┌─────┬───────┬─────┬─────┬─────┬──────┬─────┐
│name │ q0    │ q1  │ q2  │ q3  │ q4   │ q5  │
├─────┼───────┼─────┼─────┼─────┼──────┼─────┤
│  qz │  0°   │  0° │  0° │  0° │  0°  │  0° │
│  qr │  180° │  0° │  0° │  0° │  90° │  0° │
└─────┴───────┴─────┴─────┴─────┴──────┴─────┘

Rad
[0, 0.1, 0.2, 0, 0, 0]
Initial T.t
[ 2.91171999e-01  1.15647698e-18 -1.07996702e-01]
New T.t
[ 0.299172   0.001     -0.1079967]
IKsolution(q=array([ 0.003227  ,  0.09257713,  0.1710215 , -0.0261988 ,  0.03641238,
        0.02309858]), success=True, reason=None, iterations=16, residual=3.475997356554854e-11)
```

 If it is correct approach I suggest you add 1 page at wiki to show how to use it. kinematics noob like me will thank you so much.

petercorke, 2021-03-21:
This looks good, well done!  There is a big gotcha with DH parameters though, there are two flavours: standard and modified.  They all have theta, d, a, alpha.  Do you know for sure which type your robot is?  The fact that they give the order as alpha, a, d, theta makes me think it might be modified DH.  If that's the case use `RevoluteMDH` and `PrismaticMDH` to construct the links.  If you cross check the answer you get from Toolbox with an example from the other tool, that will decide it.

kstan79, 2021-03-21:
ok, im tried to put this code into AR3, the movement is not relevant, but it is good try for me. At least it can generate output. I will try to ask it is DH or modified DH. Come back to you afterwards

kstan79, 2021-03-21:
[image]
I assume I have to convert the DH parameter to m (/1000), and below is my ar3model.py
```
class AR3(DHRobot):

    def __init__(self, symbolic=False):
        if symbolic:
            import spatialmath.base.symbolic as sym
            zero = sym.zero()
            pi = sym.pi()
        else:
            from math import pi
        deg = pi / 180

        super().__init__( [
                RevoluteDH(d=0.16977, alpha=-pi / 2, a=0.0642),
                RevoluteDH(d=0, alpha=0, a=0.305),
                RevoluteDH(d=0, alpha=pi / 2, a=0),
                RevoluteDH(d=-0.22263, alpha=-pi / 2, a=0),
                RevoluteDH(d=0, alpha=pi / 2, a=0),
                RevoluteDH(d=-0.03625, alpha=0, a=0),

            ], name="AR3",manufacturer="AR3 ",)

        self.addconfiguration("qz", np.array([0, 0, 0, 0, 0, 0]))
        # horizontal along the x-axis
        self.addconfiguration("qr", np.r_[180, 0, 0, 0, 90, 0] * deg)
...
```
I notice there is no place to define `Θ`,  and I just simply define qr / qz cause it seems like not important?

kstan79, 2021-03-22:
I'd change caluclation function from `ikine_LM` to `ikine_LMS`, test few sample the result seems pretty match the original calculation result. I'm wondering what is the different of each inverse kinematic calculation. Anyhow, my problem maybe solved

petercorke, 2021-04-19:
These two differ only in the cost function.  In practice there doesn't seem to be much difference.  These two are fast than `ikine_min`

MCollins52, 2021-05-19:
> I notice there is no place to define `Θ`, and I just simply define qr / qz cause it seems like not important?

I just figured out the `Θ` parameter. Instead of this, put 'offset' like the other parameters. It should look something like this for a joint offset of 90 degrees:
> offset=pi / 2

If you print the robot's DH table, the first column should contain these offsets.

petercorke, 2021-05-19:
What exactly did you figure out?  What are you trying to do?

MCollins52, 2021-05-20:
I figured out how to put all 4 parameters from the DH table in a model. In the example where it is shown how to make a DH model, only 3 of the 4 parameters are mentioned. The fourth is offset.

petercorke, 2021-05-21:
As per my previous comment, can you please explain clearly what you are trying to do, or what the question is. What "example" are you referring to?  What kind of example/documentation would help you to minimise the issues/confusion that you have experienced.

MCollins52, 2021-05-24:
> As per my previous comment, can you please explain clearly what you are trying to do, or what the question is. What "example" are you referring to? What kind of example/documentation would help you to minimise the issues/confusion that you have experienced.

From the AR3 information supplied above, there are 4 total DH parameters. Within the DH models folder, the wiki only includes the a, d, and alpha parameters, but not Θ.

jhavl, 2021-09-13:
The DH table now includes theta (as shown in the README). To view joints offsets (since offsets are not a DH parameter) you can do `print(robot.offset)` for a list of offsets per joint.

closed 2021-09-13 by jhavl