petercorke/robotics-toolbox-python #207: Is it possible to build a whl package which contains all the pyd/dll files

asmwarrior, 2021-04-09:
I don't have Visual C++ compiler tool installed, so, when I try to install this tool through pip command, I get:

~~~~
...
Complete output (6 lines):
  running bdist_wheel
  running build
  running build_ext
  skipping 'quadprog\quadprog.cpp' Cython extension (up-to-date)
  building 'quadprog' extension
error: Microsoft Visual C++ 14.0 or greater is required. Get it with "Microsoft C++ Build Tools": https://visualstudio.microsoft.com/visual-cpp-build-tools/
~~~~

My question: is it possible to build a whl package which contains all the pyd/dll files, so I don't need to build the cpp soure file myself.

Or is it possible to make such a customized whl which contains all the dpy/dll files, because I would like to distribute this package to my students, but for now, I don't want to every students to install the Visual c++ compiler.

Thanks.

jhavl, 2021-04-09:
Thanks for bringing this to our attention, we go to great effort with the releases on [PyPi](https://pypi.org/project/roboticstoolbox-python/0.9.1/#files) so that the toolbox is pre-built for multiple operating systems and Python versions. This means that no C++ build tools are required to install the toolbox when doing `pip install roboticstoolbox-python`. 

However,  the package quadprog and its parent qpsolvers isn't under our control, and they do not pre-build wheels. Since this is a dependency of the toolbox, this breaks our _no build tools required_ approach. @petercorke and I will reconsider how we should include these in the toolbox to try and fix this.

Have a look into using the conda version of the toolbox `conda install -c conda-forge roboticstoolbox-python` where qpsolvers and quadprog should be pre-built on conda.

You can also build your own wheels of quadprog to distribute:

```
# Update and install required tools
pip install -U setuptools wheel twine

# While located in the top level directory of the package you wish to build
python setup.py sdist bdist_wheel
```
The built wheel will then be located in `dist/`. Note that this wheel is only compatible with the same operating system and major python version (eg. 3.6.x) as what it was built on. The exception is Linux where it will likely not be compatible on any other computer.

asmwarrior, 2021-04-09:
> Thanks for bringing this to our attention, we go to great effort with the releases on [PyPi](https://pypi.org/project/roboticstoolbox-python/0.9.1/#files) so that the toolbox is pre-built for [...]

Thanks for the quick response.
Unluckily, I use another Python distribution [WinPython](https://winpython.github.io/) 3.8.8.0. Since WinPython is a portable python suite, I just ask my students to unzip the file, and run everywhere. And unluckily, WinPython doesn't have pre-builds qpsolvers and quadprog installed.

> You can also build your own wheels of quadprog to distribute:
>
> ```
> # Update and install required tools
> pip install -U setuptools wheel twine
>
> # While located in the top level directory of the [...]

We mainly use Windows 7 or Windows 10, so I may build such wheel for my students, or I may ask the maintainer of WinPython to support qpsolvers and quadprog libraries as pre-installed libraries.

petercorke, 2021-04-09:
@jesse, quadprog is so far only used for the experimental IK solutions?

If yes, I think we make RTB robust to it being not present, and ask the user to install if they do something that needs it.

asmwarrior, 2021-04-11:
Hi, the good news is that WinPython will officially support the missing prebuild "the package quadprog and its parent qpsolvers".

See this issue in winpython github: [Is it possible to ship the package quadprog and its parent qpsolvers in the WinPython · Issue #955 · winpython/winpython](https://github.com/winpython/winpython/issues/955)

jhavl, 2021-09-13:
Since I don't own quadprog or qpsolvers, I can't fix this at the source. But I could submit a PR so they pre-build for PyPi and will look do do that sometime.

closed 2021-09-13 by jhavl