petercorke/robotics-toolbox-python #208: Can't open the Swift simulator in the browser (Windows 7, Python 3.8)

asmwarrior, 2021-04-09:
Hi, I `have robotics-toolbox-python` 0.9.1 successfully installed in Windows 7 with WinPython 3.8.

The mathplotlib backend works OK, while I try to switch to the `Swift` backend, I got such issue:

Here is the command line logs when I try to run it inside the IPython in side the Spyder IDE.

~~~~
from roboticstoolbox.backends.Swift import Swift  # instantiate 3D browser-based visualizer
backend = Swift()
backend.launch()            # activate it
backend.add(robot)          # add robot to the 3D scene
for qk in qt.q:             # for each joint configuration on trajectory
      robot.q = qk          # update the robot state
      backend.step()        # update visualization
      

Could not connect to the Swift simulator 

Traceback (most recent call last):

  File "<ipython-input-19-0e5c291e7847>", line 3, in <module>
    backend.launch()            # activate it

  File "E:\download\sci\Winpython64-3.8.8.0\WPy64-3880\python-3.8.8.amd64\lib\site-packages\roboticstoolbox\backends\Swift\Swift.py", line 115, in launch
    sw.start_servers(self.outq, self.inq, browser=browser)

  File "E:\download\sci\Winpython64-3.8.8.0\WPy64-3880\python-3.8.8.amd64\lib\site-packages\swift\Swift.py", line 60, in start_servers
    inq.get(timeout=10)

  File "E:\download\sci\Winpython64-3.8.8.0\WPy64-3880\python-3.8.8.amd64\lib\queue.py", line 178, in get
    raise Empty

Empty

~~~~

When I run the command, a new web page is opened by Firefox browser, with the address `http://localhost:52003/51483`

And the web contents are:

~~~~
Error response

Error code: 404

Message: File not found.

Error code explanation: HTTPStatus.NOT_FOUND - Nothing matches the given URI.
~~~~

Here is the screen shot:
[image]

Is there anything worng with my steps?

Thanks.

asmwarrior, 2021-04-10:
This looks like some issue inside the Swift Simulator.

To solve this issue, one line code in Swift Simulator should be changed, see here: https://github.com/jhavl/swift/issues/4#issuecomment-816687449

jhavl, 2021-04-14:
Hey @asmwarrior, just so I can properly debug this
Is the toolbox and Swift stored on a different drive/partition to the operating system?
And, are you running your python scripts on a different drive/partition to the operating system?

asmwarrior, 2021-04-14:
> 
> 
> Hey @asmwarrior, just so I can properly debug this
> Is the toolbox and Swift stored on a different drive/partition to the operating system?
> And, are you running your python scripts on a different drive/partition to the operating system?

My OS is under `C:/Windows/*`
MY github local copy of toolbox and swift is located in `F:/code/robotics/*`
My Python is located in `E:/download/sci/WinPython3.8.8.0/*`

jhavl, 2021-04-19:
This is fixed with this [commit](https://github.com/jhavl/swift/commit/8fd83329af29142bf9b4e5a30b4f08ca76ed6968) to swift

closed 2021-04-19 by jhavl