petercorke/robotics-toolbox-python #209: Can not start the vpython backend (Windows 7 64bit, Python 3.8.8.0)

asmwarrior, 2021-04-10:
Hi, I have those packages:
~~~~
vpython 7.6.1
swift-sim 0.8.1
roboticstoolbox-python 0.9.1
~~~~

When I try to run the example code(the code is from the official release of the robotics-toolbox-python-0.9.1 package :

~~~~
#!/usr/bin/env python
"""
@author Micah Huth
"""

import roboticstoolbox as rtb
import time
from roboticstoolbox.backends import VPython

env = VPython.VPython()  # lgtm [py/call-to-non-callable]
env.launch()

#  PUMA560
puma = rtb.models.DH.Puma560()
env.add(puma)

for i in range(1000):
    env.step(puma)
    time.sleep(0.1)
~~~~

I got the following error:

~~~~
runfile('F:/code/robotics-toolbox-python-0.9.1/examples/VPython.py', wdir='F:/code/robotics-toolbox-python-0.9.1/examples')
Traceback (most recent call last):

  File "F:\code\robotics-toolbox-python-0.9.1\examples\VPython.py", line 10, in <module>
    env = VPython.VPython()  # lgtm [py/call-to-non-callable]

AttributeError: type object 'VPython' has no attribute 'VPython'
~~~~

Any idea how to fix this issue?
Thanks.

asmwarrior, 2021-04-10:
OK, problem is solved by change the line to:

~~~~
env = VPython()
~~~~

asmwarrior, 2021-04-10:
There is another issue I see:

~~~~
In [1]: runfile('F:/code/robotics-toolbox-python-0.9.1/examples/VPython.py', wdir='F:/code/robotics-toolbox-python-0.9.1/examples')
Traceback (most recent call last):

  File "F:\code\robotics-toolbox-python-0.9.1\examples\VPython.py", line 18, in <module>
    env.step(puma)

  File "E:\download\sci\Winpython64-3.8.8.0\WPy64-3880\python-3.8.8.amd64\lib\site-packages\roboticstoolbox\backends\VPython\VPython.py", line 215, in step
    sleep(dt)

TypeError: an integer is required (got type Puma560)
~~~~

The line is: `env.step(puma)`.

I think some interface is changed, but this example code is not updated yet.

asmwarrior, 2021-04-10:
I uninstall all the pipy version of the three packages:
~~~~
robotics-toolbox-python
spatialmath-python
swift
~~~~

And install all the github versions, this removes a lot of errors when I try to run the codes in example folder. (To use swift-sim under Windows, I have to use a patch here: https://github.com/jhavl/swift/issues/4#issuecomment-817119736 )

Now, I still see some errors:

When run the file:  `\robotics-toolbox-python\roboticstoolbox\examples\vpython-puma.py`, I get such error:

~~~~
In [1]: runfile('E:/code/robotics/robotics-toolbox-python/roboticstoolbox/examples/vpython-puma.py', wdir='E:/code/robotics/robotics-toolbox-python/roboticstoolbox/examples')
Traceback (most recent call last):

  File "E:\code\robotics\robotics-toolbox-python\roboticstoolbox\examples\vpython-puma.py", line 22, in <module>
    env.step(0.1)

  File "e:\code\robotics\robotics-toolbox-python\roboticstoolbox\backends\VPython\VPython.py", line 204, in step
    graphical_dh_robot.fkine_and_set(q)

  File "e:\code\robotics\robotics-toolbox-python\roboticstoolbox\backends\VPython\graphicalrobot.py", line 802, in fkine_and_set
    poses = self.robot.fkine_all(joint_angles)

  File "e:\code\robotics\robotics-toolbox-python\roboticstoolbox\robot\DHRobot.py", line 902, in fkine_all
    Tall.append(Tj)

  File "e:\code\robotics\spatialmath-python\spatialmath\baseposelist.py", line 368, in append
    raise ValueError("can't append a multivalued instance - use extend")

ValueError: can't append a multivalued instance - use extend

In [2]: 
~~~~

Any ideas how to fix this issue? 
Thanks.

asmwarrior, 2021-04-10:
Another issue I see when I use the github version of the "robotics-toolbox-python, spatialmath-python, swift" is that when I try to run a minimal code in the home page:

~~~~
import roboticstoolbox as rp

panda = rp.models.Panda()
panda.plot(q=panda.qr)
~~~~

Now, I get such error:
~~~~
In [1]: runfile('E:/code/robotics/robotics-toolbox-python/roboticstoolbox/examples/mytest.py', wdir='E:/code/robotics/robotics-toolbox-python/roboticstoolbox/examples')
Traceback (most recent call last):

  File "E:\code\robotics\robotics-toolbox-python\roboticstoolbox\examples\mytest.py", line 4, in <module>
    panda.plot(q=panda.qr)

  File "e:\code\robotics\robotics-toolbox-python\roboticstoolbox\robot\Robot.py", line 1151, in plot
    env = self._get_graphical_backend(backend)

  File "e:\code\robotics\robotics-toolbox-python\roboticstoolbox\robot\Robot.py", line 1050, in _get_graphical_backend
    from roboticstoolbox.backends.Swift import Swift

ModuleNotFoundError: No module named 'roboticstoolbox.backends.Swift'

In [2]: 
~~~~

asmwarrior, 2021-04-10:
> Now, I still see some errors:
>
> When run the file: `\robotics-toolbox-python\roboticstoolbox\examples\vpython-puma.py`, I get such error:
>
> ```
> In [1]: [...]

I just debugged this problem, and I notice that some times, 

https://github.com/petercorke/robotics-toolbox-python/blob/5468a894c0127a24008bddc0e4ba3fc339d8a507/roboticstoolbox/robot/DHRobot.py#L846

In this function, the ` Tj = self._base`, I see that the `_base` is an array, not a single `SE(3)`, so to solve this issue, I have such changes:

~~~~
 roboticstoolbox/robot/DHRobot.py | 2 +-
 1 file changed, 1 insertion(+), 1 deletion(-)

diff --git a/roboticstoolbox/robot/DHRobot.py b/roboticstoolbox/robot/DHRobot.py
index 96fff873..34ef6dc5 100644
--- a/roboticstoolbox/robot/DHRobot.py
+++ b/roboticstoolbox/robot/DHRobot.py
@@ -882,7 +882,7 @@ class DHRobot(Robot):
         q = self._getq(q)
 
         if self._base is not None:
-            Tj = self._base
+            Tj = self._base[0]
         else:
             Tj = SE3()
         first = True

~~~~

Now, which the above changes, the crash issue is gone. But I still don't know why the `_base` is an array.

Here is the test code for this issue:

~~~~
import roboticstoolbox as rtb
import time

from roboticstoolbox.backends.VPython import VPython

env = VPython()

env.launch()

#  PANDA
panda = rtb.models.DH.Panda()
env.add(panda)

time.sleep(3)
env.step(id = panda, q = panda.qr, dt = 0.1)

time.sleep(3)
env.step(id = panda, q = panda.qz, dt = 0.1)

~~~~

The code is mainly based from the file: `robotics-toolbox-python\roboticstoolbox\examples\vpython-puma.py`, but should change some codes, such as `rp` should be replaced by `rtb`.

micah-huth, 2021-04-15:
Hey, sorry I'm a bit behind in getting to this.

> 
> 
> I uninstall all the pipy version of the three packages:
> 
> ```
> robotics-toolbox-python
> spatialmath-python
> swift
> ```
> 
> And install all the github versions, this removes a lot of errors when I try to run the codes in example folder. (To use swift-sim under Windows, I have to use a patch here: [jhavl/swift#4 (comment)](https://github.com/jhavl/swift/issues/4#issuecomment-817119736) )
> 
> Now, I still see some errors:
> 
> When run the file: `\robotics-toolbox-python\roboticstoolbox\examples\vpython-puma.py`, I get such error:
> 
> ```
> In [1]: runfile('E:/code/robotics/robotics-toolbox-python/roboticstoolbox/examples/vpython-puma.py', wdir='E:/code/robotics/robotics-toolbox-python/roboticstoolbox/examples')
> Traceback (most recent call last):
> 
>   File "E:\code\robotics\robotics-toolbox-python\roboticstoolbox\examples\vpython-puma.py", line 22, in <module>
>     env.step(0.1)
> 
>   File "e:\code\robotics\robotics-toolbox-python\roboticstoolbox\backends\VPython\VPython.py", line 204, in step
>     graphical_dh_robot.fkine_and_set(q)
> 
>   File "e:\code\robotics\robotics-toolbox-python\roboticstoolbox\backends\VPython\graphicalrobot.py", line 802, in fkine_and_set
>     poses = self.robot.fkine_all(joint_angles)
> 
>   File "e:\code\robotics\robotics-toolbox-python\roboticstoolbox\robot\DHRobot.py", line 902, in fkine_all
>     Tall.append(Tj)
> 
>   File "e:\code\robotics\spatialmath-python\spatialmath\baseposelist.py", line 368, in append
>     raise ValueError("can't append a multivalued instance - use extend")
> 
> ValueError: can't append a multivalued instance - use extend
> 
> 
> In [2]: 
> ```
> 
> Any ideas how to fix this issue?
> Thanks.

In relation to this, I tested the code in the presumed commit stage you were at (based upon time of report, I also got the same error), and at the latest stage as of this reply. It has since been fixed since your report, as the latest commit executes without error.

> Another issue I see when I use the github version of the "robotics-toolbox-python, spatialmath-python, swift" is that when I try to run a minimal code in the home page:
>
> ```
> import roboticstoolbox [...]

This error is not VPython specific. When no backend is supplied, the default is used depending on the application (currently: pyplot for DH robots, pyplot2 for ERobots2, Swift for everything else). As you are using the ERobot version of Panda, it will try to launch in Swift.

https://github.com/jhavl/swift/issues/4#issuecomment-816610553
As for this issue, I'm guessing this is no longer an issue, as you have posted these error reports above?

asmwarrior, 2021-04-15:
> > Another issue I see when I use the github version of the "robotics-toolbox-python, spatialmath-python, swift" is that when I try to run a minimal code in the home page:
> > ```
> > import roboticstoolbox as rp
> > 
> > panda = rp.models.Panda()
> > panda.plot(q=panda.qr)
> > ```
> > 
> > 
> > Now, I get such error:
> > ```
> > In [1]: runfile('E:/code/robotics/robotics-toolbox-python/roboticstoolbox/examples/mytest.py', wdir='E:/code/robotics/robotics-toolbox-python/roboticstoolbox/examples')
> > Traceback (most recent call last):
> > 
> >   File "E:\code\robotics\robotics-toolbox-python\roboticstoolbox\examples\mytest.py", line 4, in <module>
> >     panda.plot(q=panda.qr)
> > 
> >   File "e:\code\robotics\robotics-toolbox-python\roboticstoolbox\robot\Robot.py", line 1151, in plot
> >     env = self._get_graphical_backend(backend)
> > 
> >   File "e:\code\robotics\robotics-toolbox-python\roboticstoolbox\robot\Robot.py", line 1050, in _get_graphical_backend
> >     from roboticstoolbox.backends.Swift import Swift
> > 
> > ModuleNotFoundError: No module named 'roboticstoolbox.backends.Swift'
> > 
> > 
> > In [2]: 
> > ```
> 
> This error is not VPython specific. When no backend is supplied, the default is used depending on the application (currently: pyplot for DH robots, pyplot2 for ERobots2, Swift for everything else). As you are using the ERobot version of Panda, it will try to launch in Swift.
> 
> [jhavl/swift#4 (comment)](https://github.com/jhavl/swift/issues/4#issuecomment-816610553)
> As for this issue, I'm guessing this is no longer an issue, as you have posted these error reports above?

See my post here: https://github.com/jhavl/swift/issues/4#issuecomment-820017336
There are still some code need to be fixed in the robotics toolbox.

asmwarrior, 2021-04-15:
> Hey, sorry I'm a bit behind in getting to this.
>
>
> In relation to this, I tested the code in the presumed commit stage you were at (based upon time of report, I also got the same error), and at the [...]

When I revert my local change of this:
~~~~
diff --git a/roboticstoolbox/robot/DHRobot.py b/roboticstoolbox/robot/DHRobot.py
index 133f1712..55f38515 100644
--- a/roboticstoolbox/robot/DHRobot.py
+++ b/roboticstoolbox/robot/DHRobot.py
@@ -979,7 +979,7 @@ class DHRobot(Robot):
         q = self._getq(q)
 
         if self._base is not None:
-            Tj = self._base
+            Tj = self._base[0]
         else:
             Tj = SE3()
         first = True

~~~~
try to run a simplified `examples/vpython-puma.py` file:

~~~~
import roboticstoolbox as rtb
import time

from roboticstoolbox.backends.VPython import VPython

env = VPython()

env.launch()

panda = rtb.models.DH.Panda()
env.add(panda)

time.sleep(3)
env.step(id = panda, q = panda.qr, dt = 0.1)

time.sleep(3)
env.step(id = panda, q = panda.qz, dt = 0.1)
~~~~

I don't see the error again, so I guess it was fixed in github master.

In-fact, I set a breakpoint in the line: `Tj = self._base` in my above diff file(in line 983 in DHRobot.py), and I see it never get reached this breakpoint, I'm not sure which commit fixed this issue. 

Thanks.

micah-huth, 2021-04-16:
Closing this VPython issue thread, as the VPython issue(s) have been fixed.

closed 2021-04-16 by micah-huth