petercorke/robotics-toolbox-python #210: error running examples/puma_jtraj.py (Windows 7 64bit, Python 3.8.8.0)

asmwarrior, 2021-04-10:
Hi, I have those packages:

~~~~
vpython 7.6.1
swift-sim 0.8.1
roboticstoolbox-python 0.9.1
spatialmath-python 0.9.5
~~~~

When I try to run the file "examples/puma_jtraj.py", which I extract from the official release: [Source code (zip)](https://github.com/petercorke/robotics-toolbox-python/archive/refs/tags/v0.9.1.zip)

I got such error:

~~~~
In [2]: runfile('F:/code/robotics-toolbox-python-0.9.1/examples/puma_jtraj.py', wdir='F:/code/robotics-toolbox-python-0.9.1/examples')
┏━━━━┳━━━━━━━━━━━━━━━━━━━━┳━━━━━━━━┳━━━━━━━━┳━━━━━━━━━┳━━━━━━━━┓
┃θⱼ  ┃         dⱼ         ┃   aⱼ   ┃   ⍺ⱼ   ┃   q⁻    ┃   q⁺   ┃
┣━━━━╋━━━━━━━━━━━━━━━━━━━━╋━━━━━━━━╋━━━━━━━━╋━━━━━━━━━╋━━━━━━━━┫
┃ q1 ┃ 0.6718299999999999 ┃      0 ┃  90.0° ┃ -160.0° ┃ 160.0° ┃
┃ q2 ┃                  0 ┃ 0.4318 ┃   0.0° ┃ -110.0° ┃ 110.0° ┃
┃ q3 ┃            0.15005 ┃ 0.0203 ┃ -90.0° ┃ -135.0° ┃ 135.0° ┃
┃ q4 ┃             0.4318 ┃      0 ┃  90.0° ┃ -266.0° ┃ 266.0° ┃
┃ q5 ┃                  0 ┃      0 ┃ -90.0° ┃ -100.0° ┃ 100.0° ┃
┃ q6 ┃                  0 ┃      0 ┃   0.0° ┃ -266.0° ┃ 266.0° ┃
┗━━━━┻━━━━━━━━━━━━━━━━━━━━┻━━━━━━━━┻━━━━━━━━┻━━━━━━━━━┻━━━━━━━━┛

┌─────┬─────┬──────┬───────┬─────┬──────┬─────┐
│name │ q0  │ q1   │ q2    │ q3  │ q4   │ q5  │
├─────┼─────┼──────┼───────┼─────┼──────┼─────┤
│  qz │  0° │  0°  │  0°   │  0° │  0°  │  0° │
│  qr │  0° │  90° │ -90°  │  0° │  0°  │  0° │
│  qs │  0° │  0°  │ -90°  │  0° │  0°  │  0° │
│  qn │  0° │  45° │  180° │  0° │  45° │  0° │
└─────┴─────┴──────┴───────┴─────┴──────┴─────┘

Traceback (most recent call last):

  File "F:\code\robotics-toolbox-python-0.9.1\examples\puma_jtraj.py", line 44, in <module>
    for q in qt.q:

AttributeError: 'Trajectory' object has no attribute 'q'
~~~~

I see that the matplotlib window is already opened, and the Puma 560 robot is already shown, but in the line 44:

~~~~
backend.launch()
backend.add(robot)
for q in qt.q:   #  -----------------------this line
	robot.q = q
	backend.step(0.1)
~~~~

I got this error, any ideas how to fix this error?
Thanks.

umerhuzaifa, 2021-05-03:
Try by replacing 'qt.q' with 'qt.y'.

asmwarrior, 2021-05-04:
> 
> 
> Try by replacing 'qt.q' with 'qt.y'.

Hi, thanks for the reply. In-fact, I don't know what does the `qt.y` mean.

Anyway, I just test the code with the github version of robotics-toolbox-python and other libraries(vpython, swift...), and I don't see the errors now.

So, I think this issue is already fixed in the github version now, I will close it.

closed 2021-05-04 by asmwarrior