petercorke/robotics-toolbox-python #212: rtb-data big package

stonebig, 2021-04-10:
would it be possible to make the 'rtb-data' package an optional dependency ?

asmwarrior, 2021-04-11:
[stonebig](https://github.com/stonebig) is the maintainer of the [WinPython](https://winpython.github.io/), I think he may consider adding this package to WinPython.

petercorke, 2021-04-12:
It is big, that's why we split it out.  If there was a way that a user could elect to not include it that would be great.  I would like the default action to be including it.  Do you know how to do this?

stonebig, 2021-04-12:
Seaborn doesn't include its small datasets in the package, intake or torch seems about similar, with bigger datasets.

If it's passive datas and splittable  per robots dataset, it looks the natural way.
 
If you create the rtb-data dependency as an extra option package, it would be a first step

You would recommand to load package as robotic[all]

jhavl, 2021-04-13:
We want to make usage as easy as possible for the majority of use cases for the toolbox. So including rtb-data by default will hold to this menatality. 

However, I have investigated some other solutions to accomodate your request. Unfortunatly, setuptools is rather inflexible regarding re-defining install-requires in optional requirements. There is some prgress on this front [here](https://github.com/pypa/setuptools/issues/1139) so it may be possible in the future.

In the meantime I have made a minimal version of rtb-data with is only 2mb. It must be installed before the toolbox:
```
pip install rtb-data==0.0.1
pip install roboticstoolbox-python
```

closed 2021-04-13 by jhavl