petercorke/robotics-toolbox-python #213: alpha parameter of DHRobot does not show real alpha of joints

OrionHeart, 2021-04-10:
Basically DHRobot.alpha does not show the real alpha value of joints, like you can see in the following picture where I have +-pi/2 in some of the joints but model.alpha is all zeros.

[image]

This is more like an asthetic bug, because everything seems to be calculated fine with the real alpha value per joint, but in case you want to gather alpha values for some other thing, then you can't do "robot.alpha" and you are forced to do " [robot.links[j].alpha for j in range(robot.n)] " or something like that

I'm using latest toolbox v0.9.1.

btw, I love to have this toolbox now on python, I've using the matlab evrsion for quite some time now. Good joob!

Sergio

petercorke, 2021-04-12:
Thx, I'll add this one to the list, I think it was supposed to do what you want...

petercorke, 2021-04-19:
Fixed

closed 2021-04-19 by petercorke