petercorke/robotics-toolbox-python #215: rot2jac is missing from the most recent versions

dimitriSchreiber, 2021-04-11:
**problem description**
rot2jac is missing from the most recent versions. Therefore, the mmc.py example fails to run.

Traceback (most recent call last):
  File "mmc.py", line 7, in <module>
    import roboticstoolbox as rtb
  File "/home/dimitri/Desktop/work/robotics-toolbox-python/roboticstoolbox/__init__.py", line 2, in <module>
    from roboticstoolbox.robot import *
  File "/home/dimitri/Desktop/work/robotics-toolbox-python/roboticstoolbox/robot/__init__.py", line 1, in <module>
    from roboticstoolbox.robot.Robot import Robot
  File "/home/dimitri/Desktop/work/robotics-toolbox-python/roboticstoolbox/robot/Robot.py", line 22, in <module>
    from roboticstoolbox.robot.Dynamics import DynamicsMixin
  File "/home/dimitri/Desktop/work/robotics-toolbox-python/roboticstoolbox/robot/Dynamics.py", line 15, in <module>
    from spatialmath.base import \
ImportError: cannot import name 'rot2jac'

**Environment**
 - Ubuntu 18.04
- Python 3.6.9

petercorke, 2021-04-12:
Sorry, there's a bit of work going on around this.  Give us a few days to put things back together.

oridong, 2021-04-14:
I also join waiting this.

petercorke, 2021-04-19:
This function is actually in the spatial math toolbox, version pushed to PyPI yesterday has this fixed.  There's a new function `angvelxform` which aims to be a bit more general than `rot2jac`.  Still haven't got exponential coordinates working.

closed 2021-04-19 by petercorke