petercorke/robotics-toolbox-python #217: I would suggest some code/development guidlines

asmwarrior, 2021-04-16:
When using the github version of this library and the associated library, I always see the examples code get running errors here or there. It's a bit mass to debug here and there, even I don't understand much part of this library.

When viewing the commit log messages, some of them are quite simple, and a bit hard to understand.

Can you set a rule that for each commit, it should pass every tests or at least the example code should work fine.
For the commit message, it should have detailed descriptions.

When the commit is not stable, it should not be merged into the master branch.

Any ideas?

Thanks.

petercorke, 2021-04-17:
We're still experimenting. I'm not as disciplined as I should be and probably should work more on branches.  The PyPi versions are coherent.  We're working on tidying things up over coming weeks.

asmwarrior, 2021-04-17:
OK, thanks.

I will report any bugs/errors when I see in the recent github master version.

petercorke, 2021-04-19:
Thanks.  We'll get things back to passing then look at our workflow.  We are also planning to move to black which will cause big changes throughout.

closed 2021-04-19 by petercorke