petercorke/robotics-toolbox-python #220: 'Mesh' object has no attribute '_sq'

oridong, 2021-04-20:
win10 python3.8.9

'Mesh' object has no attribute '_sq'

error parsing URDF file D:\0ws_python\0MyPackages\rtb_data-0.9.1-py3.8\rtbdata\xacro\franka_description\robots\panda_arm_hand.urdf.xacro

Giessen, 2021-04-20:
Same problem here....

jhavl, 2021-04-20:
Try doing `pip install -U spatialgeometry` and check out [common issues](https://github.com/petercorke/robotics-toolbox-python/wiki/Common-Issues) too see if any points there help

Giessen, 2021-04-20:
I try uninstall spatialgeometry and reinstall from source, it works! Thanks

closed 2021-04-20 by jhavl

oridong, 2021-04-21:
> I try uninstall spatialgeometry and reinstall from source, it works! Thanks
first I uninstalled spatialgeometry.
I dowload it from https://github.com/jhavl/spatialgeometry, and python setup.py install , but the problem still exists.
Then  I  uninstalled spatialgeometry and `pip install -U spatialgeometry` , but the problem also exists.
How did you do ?

jhavl, 2021-04-21:
If you do `pip list` what version of spatialgeometry do you see?

oridong, 2021-04-21:
> If you do `pip list` what version of spatialgeometry do you see?
roboticstoolbox-python 0.9.1
spatialgeometry        0.1.0
spatialmath-python     0.9.6

jhavl, 2021-04-21:
spatialgeometry should have been v0.1.1. The PyPI release from yesterday didn't work properly so your `pip install -U spatialgeometry` didn't update anything. I've just re-released it with version 0.1.2 and so now if you do `pip install -U spatialgeometry` you should end up with version 0.1.2 and everything will work.

oridong, 2021-04-21:
> spatialgeometry should have been v0.1.1. The PyPI release from yesterday didn't work properly so your `pip install -U spatialgeometry` didn't update anything. I've just re-released it with version [...]

Ok, this problem is solved.