petercorke/robotics-toolbox-python #221: No module named 'roboticstoolbox.backends.Swift'  No module named 'swift.SwiftRoute'

oridong, 2021-04-21:
when
```
from roboticstoolbox.backends.Swift import Swift  # instantiate 3D browser-based visualizer
backend = Swift()
backend.launch()            # activate it
backend.add(robot)          # add robot to the 3D scene
for qk in qt.q:             # for each joint configuration on trajectory
      robot.q = qk          # update the robot state
      backend.step()        # update visualization
```
No module named 'roboticstoolbox.backends.Swift'

when I use 
`from roboticstoolbox.backends.swift import Swift  # instantiate 3D browser-based visualizer1

No module named 'swift.SwiftRoute'

```
site-packages\roboticstoolbox_python-0.9.1-py3.8-win-amd64.egg\roboticstoolbox\backends\swift\__init__.py", line 1, in <module>
    from swift.SwiftRoute import SwiftServer, SwiftSocket, start_servers
ModuleNotFoundError: No module named 'swift.SwiftRoute'
```

asmwarrior, 2021-04-21:
I think this is already reported, and now it is fixed in the github version.

See here:

https://github.com/jhavl/swift/issues/4#issuecomment-820017336

oridong, 2021-04-21:
> I think this is already reported, and now it is fixed in the github version.
> 
> See here:
> 
> [jhavl/swift#4 (comment)](https://github.com/jhavl/swift/issues/4#issuecomment-820017336)

I got it form git clone

closed 2021-04-21 by oridong