petercorke/robotics-toolbox-python #222: can't print the robot if symbolic is enabled

asmwarrior, 2021-04-21:
Hi, I'm using the github version of both `robotics-toolbox-python`, `spatialgeometry`, `spatialmath-python`, `swift`.

Now, run the such code:

~~~~
import roboticstoolbox as rtb
puma = rtb.models.DH.Puma560(symbolic=True)
print(puma)
~~~~

I got such error:

~~~~
runfile('E:/code/robotics/robotics-toolbox-python/tests/test_1.py', wdir='E:/code/robotics/robotics-toolbox-python/tests')
Traceback (most recent call last):

  File "E:\code\robotics\robotics-toolbox-python\tests\test_1.py", line 14, in <module>
    print(puma)

  File "e:\code\robotics\robotics-toolbox-python\roboticstoolbox\robot\DHRobot.py", line 252, in __str__
    s += self.configurations_str()

  File "e:\code\robotics\robotics-toolbox-python\roboticstoolbox\robot\Robot.py", line 408, in configurations_str
    qlist.append(angle(q[j], "{: .3g}"))

  File "e:\code\robotics\robotics-toolbox-python\roboticstoolbox\robot\Robot.py", line 389, in angle
    return fmt.format(theta * deg) + "\u00b0"

TypeError: unsupported format string passed to Mul.__format__

~~~~

Can you fix this? Thanks.

asmwarrior, 2021-04-21:
To workaround this, I have to always use the `else` clause. such as:

~~~~
 roboticstoolbox/robot/Robot.py | 2 +-
 1 file changed, 1 insertion(+), 1 deletion(-)

diff --git a/roboticstoolbox/robot/Robot.py b/roboticstoolbox/robot/Robot.py
index c74311b6..770f0755 100644
--- a/roboticstoolbox/robot/Robot.py
+++ b/roboticstoolbox/robot/Robot.py
@@ -385,7 +385,7 @@ class Robot(ABC, DynamicsMixin, IKMixin):
         # TODO: factor this out of DHRobot
         def angle(theta, fmt=None):
 
-            if fmt is not None:
+            if False: #fmt is not None:
                 return fmt.format(theta * deg) + "\u00b0"
             else:  # pragma nocover
                 return str(theta * deg) + "\u00b0"
~~~~

petercorke, 2021-05-03:
Yeah, this needs a bit of a rethink about a better way to format numerical and symbolic data.  It's on the list, but I'll tag it here.

closed 2021-09-13 by jhavl