petercorke/robotics-toolbox-python #224: Is it possible to enable the printer of the symbolic output of matrix, such as SE3

asmwarrior, 2021-04-21:
Hi, the python's sympy has rich printer/renders to let user choose, see this link: [Printing — SymPy 1.8 documentation](https://docs.sympy.org/latest/tutorial/printing.html)

So, I would suggest if `robotics-toolbox-python` library could support this:

1, it could output the raw latex text string, so that user can copy the output text string, and paste to their `.tex` document.

2, it could select the printer/render through the sympy, and when user print the matrix, a nicely latex or other render based output can be achieved.

Thanks.

petercorke, 2021-05-03:
You can convert any RTB or SMTB type to SymPy if you want.  Useful for LaTeX generation as you say or codegen.  I want some level of SymPy formatting within the toolbox, but for complex stuff it makes sense to go to SymPy directly.

Are you talking about `SE3` matrices or other classes?

asmwarrior, 2021-05-03:
> Are you talking about `SE3` matrices or other classes?

Yes, I expect `SE3` class could directly generates `latex` format output.

petercorke, 2021-05-04:
You can turn an `SE3` into latex pretty easily, and if you wanted you could wrap the functionality.  Not sure it's a common enough use case to warrant building it in.

```
(rtb) >>> T = SE3.Rx(a)

(rtb) >>> T
Out[12]: 
   1         0         0         0         
   0        cos(theta)   -sin(theta)   0         
   0        sin(theta)   cos(theta)    0         
   0         0         0         1         

(rtb) >>> Matrix(T.A)
Out[13]: 
Matrix([
[1,          0,           0, 0],
[0, cos(theta), -sin(theta), 0],
[0, sin(theta),  cos(theta), 0],
[0,          0,           0, 1]])

(rtb) >>> latex(Matrix(T.A))
Out[14]: '\\left[\\begin{matrix}1 & 0 & 0 & 0\\\\0 & \\cos{\\left(\\theta \\right)} & - \\sin{\\left(\\theta \\right)} & 0\\\\0 & \\sin{\\left(\\theta \\right)} & \\cos{\\left(\\theta \\right)} & 0\\\\0 & 0 & 0 & 1\\end{matrix}\\right]'

```

closed 2021-05-04 by petercorke

asmwarrior, 2021-05-04:
OK, thanks for the help!
Here is the simple test code I can show latex rendered output:

~~~~
import sympy
sympy.init_printing(use_latex='svg', forecolor='black')

from spatialmath import *
import spatialmath.base.symbolic as sym
theta = sym.symbol('theta')

T = SE3.Rx(theta)

a = sympy.Matrix(T.A)

display(a)

print(sympy.latex(a))
~~~~

And this is the output from Spyder IDE when run this code:

![latex-render](https://user-images.githubusercontent.com/561818/116992812-3165d400-ad09-11eb-9da0-3cee7f0fdf78.png)

petercorke, 2021-05-05:
OK, that looks pretty nice!