petercorke/robotics-toolbox-python #226: fkine_all doesnot work

Giessen, 2021-04-21:
version:
robotics-toolbox-python
branch Master
commit dd7cd6493400033f48101b4998c9d17f41f449e6

reproduce:
--------------
import roboticstoolbox as rtb
robot = rtb.models.URDF.Panda()
robot.fkine_all(robot.qz)
->
Traceback (most recent call last):
  File "C:\Users\sun\Downloads\work\Python&&RS\python_gui\venv37\lib\site-packages\IPython\core\interactiveshell.py", line 3437, in run_code
    exec(code_obj, self.user_global_ns, self.user_ns)
  File "<ipython-input-5-100047a23859>", line 1, in <module>
    robot.fkine_all(robot.qz)
  File "C:\Users\sun\Downloads\work\Python&&RS\python_gui\venv37\lib\site-packages\roboticstoolbox\robot\ERobot.py", line 684, in fkine_all
    recurse(linkframes, Tbase, q, self.links[0])
  File "C:\Users\sun\Downloads\work\Python&&RS\python_gui\venv37\lib\site-packages\roboticstoolbox\robot\ERobot.py", line 669, in recurse
    Tall[link.number] = T
  File "C:\Users\sun\Downloads\work\Python&&RS\python_gui\venv37\lib\site-packages\spatialmath\baseposelist.py", line 328, in __setitem__
    self.data[i] = value.A
IndexError: list assignment index out of range

petercorke, 2021-05-03:
Interesting.  It works for the other Panda models, ETS and DH.  Something about the kinematic structure of this URDF model.  Not immediately clear what's going on, so I've tagged it for later.  There's been a lot of flux around `fkine_all` and I had hoped I wouldn't have to look at it again for a while...

petercorke, 2021-05-09:
Fixed in 0.10.1 and on GH master