petercorke/robotics-toolbox-python #229: examples\neo.py failed to run with the github version of robotics-toolbox-python and its friend libraries

asmwarrior, 2021-04-22:
Hi, I'm using the github version of both `robotics-toolbox-python`, `spatialgeometry`, `spatialmath-python`, `swift`.
The `pybullet` is also installed by the pip command.

Here is the error:

~~~~
Traceback (most recent call last):
  File "E:\code\robotics\robotics-toolbox-python\roboticstoolbox\examples\neo.py", line 149, in <module>
    step()
  File "E:\code\robotics\robotics-toolbox-python\roboticstoolbox\examples\neo.py", line 110, in step
    c_Ain, c_bin = panda.link_collision_damper(
  File "E:\code\robotics\robotics-toolbox-python\roboticstoolbox\robot\ERobot.py", line 2046, in link_collision_damper
    l_Ain, l_bin, d, wTcp = indiv_calculation(
  File "E:\code\robotics\robotics-toolbox-python\roboticstoolbox\robot\ERobot.py", line 2019, in indiv_calculation
    d, wTlp, wTcp = link_col.closest_point(shape, di)
  File "e:\code\robotics\spatialgeometry\spatialgeometry\CollisionShape.py", line 68, in wrapper_check_pyb
    return func(*args, **kwargs)
  File "e:\code\robotics\spatialgeometry\spatialgeometry\CollisionShape.py", line 130, in closest_point
    self._init_pob()
  File "e:\code\robotics\spatialgeometry\spatialgeometry\CollisionShape.py", line 271, in _init_pob
    col = p.createCollisionShape(
AttributeError: 'NoneType' object has no attribute 'createCollisionShape'
~~~~

It looks like `p` is NoneType.  I see that in the `CollisionShape.py`, `p` is a global variable, it should be initialized by the `_import_pyb()` function, but I'm not sure it get called.

Any ideas how to fix this issue?

Thanks.

jhavl, 2021-05-06:
That's interesting, I believe the error is that you don't have `pybullet` installed (we use pybullet for collision detection), but you should have gotten a nice message asking you to install it instead of this.

asmwarrior, 2021-05-08:
Hi, thanks for the reply.

I'm sorry that I did forget to install the `pybullet` library. 

After install `pybullet`, the `neo.py` works without issue now.

Sorry for the noise, I will close this issue.