petercorke/robotics-toolbox-python #236: robot.rne function does not work for robot created by URDF

Adineh94, 2021-05-04:
Hi,
I have created two robots from build-in functions, as following. However, I am not able to get "inertia" matrix for robot which has been created from URDF, and I have received this error:

  **File "C:\Users\moham\miniconda3\envs\rtb\lib\site-packages\roboticstoolbox\robot\ERobot.py", line 2102, in rne
    s[j] = link.v.s
AttributeError: 'NoneType' object has no attribute 's'**
#############Code#########################
import roboticstoolbox as rtb
panda = rtb.models.URDF.Panda()
puma = rtb.models.DH.Puma560()

print(puma.inertia(puma.qr))
print(panda.inertia(panda.qr))

petercorke, 2021-05-05:
That URDF model has no dynamic parameters set

```
(rtb) >>> panda.dynamics()

┌────────────┬────┬────────────┬────────────────────────┬────┬────┬────────┬────┐
│     j      │ m  │     r      │           I            │ Jm │ B  │   Tc   │ G  │
├────────────┼────┼────────────┼────────────────────────┼────┼────┼────────┼────┤
│panda_link0 │  0 │  0,  0,  0 │  0,  0,  0,  0,  0,  0 │  0 │  0 │  0,  0 │  0 │
│panda_link1 │  0 │  0,  0,  0 │  0,  0,  0,  0,  0,  0 │  0 │  0 │  0,  0 │  0 │
│panda_link2 │  0 │  0,  0,  0 │  0,  0,  0,  0,  0,  0 │  0 │  0 │  0,  0 │  0 │
│panda_link3 │  0 │  0,  0,  0 │  0,  0,  0,  0,  0,  0 │  0 │  0 │  0,  0 │  0 │
│panda_link4 │  0 │  0,  0,  0 │  0,  0,  0,  0,  0,  0 │  0 │  0 │  0,  0 │  0 │
│panda_link5 │  0 │  0,  0,  0 │  0,  0,  0,  0,  0,  0 │  0 │  0 │  0,  0 │  0 │
│panda_link6 │  0 │  0,  0,  0 │  0,  0,  0,  0,  0,  0 │  0 │  0 │  0,  0 │  0 │
│panda_link7 │  0 │  0,  0,  0 │  0,  0,  0,  0,  0,  0 │  0 │  0 │  0,  0 │  0 │
│panda_link8 │  0 │  0,  0,  0 │  0,  0,  0,  0,  0,  0 │  0 │  0 │  0,  0 │  0 │
└────────────┴────┴────────────┴────────────────────────┴────┴────┴────────┴────┘
```

but that shouldn't lead to the error you report (which I also get). None of our shipped Panda models have dynamic parameters at the moment, but the MATLAB DH model does include inertial parameters.

I'll log this a bug.

petercorke, 2021-05-09:
On a quick look this is due to static joints, ie. two links with a constant transform between them.  I will need to dig into the originals papers again to sort this out, but that's not likely to happen soon.

LukasDriller, 2021-05-16:
Hi there, 

I had the same issue with the static joints. Thus, I have rewritten the constructor an the RNE method from the ERobot class, which seems to cause the problem for me. The issue is caused by choosing `n` as the amount of joints in the robot where if should actually be evaluated for every link. `s` should only be computed for real joint links. In rne forward path respectively all values of s, q, dq and dqq should be set to zero. In backward pass only Qs of joints need to be computed. I only did some brief plausibility checks on the code but it might help on solving the issue. Below you might find my code. I would be happy to have some feedback if it helped. 

Kind Regards 
Lukas

```
#!/usr/bin/env python3
"""
@author: Lukas Driller
"""

import numpy as np
from spatialmath import SE3, SE2
from spatialmath.base.argcheck import islistof

from roboticstoolbox.robot.ERobot import ERobot
from roboticstoolbox.robot.ELink import ELink
from roboticstoolbox.robot.ETS import ETS

from spatialmath import (
    SpatialAcceleration,
    SpatialVelocity,
    SpatialInertia,
    SpatialForce,
)

import fknm

class DynamicERobot(ERobot):
    """
    Dynamic ERobot implementation of robotics toolbox, as current version does not support handing in dynamics values
    DynamicERobot() is an abstract class which allows for modelling of a Elementary Transfer Sequence (ETS) Robot.
    The class overrides the constructor of the ERobot class in the robotics toolbox and forwards dynamic parameters.
    Additionally, the rne function is overwritten to treat models with no joint at end appropriately.
    """

    def __init__(self, arg, m_list, r_list, I_list, **kwargs):
        """
        Constructor override of ERobot taking ETS and lists of dynamic parameters to create links. These are handed to
        the parent constructor. m, r and I values are handed in as lists of 1-value, 3-value and 6-value
        respectively and should contain as many values as links are present in robot. Convention:
            - m mass of link
            - distance of ith origin to center of mass [x,y,z] in link reference frame
            - inertia tensor of link with respect to center of mass I = [L_xx, L_yy, L_zz, L_xy, L_yz, L_xz]

        :param m_list: list of masses one for each joint
        :param r_list: list of center of mass position with respect to link frame
        :param I_list: list of inertia of link with respect to COM
        """
        if isinstance(arg, ETS):
            # we're passed an ETS string
            links = []
            # chop it up into segments, a link frame after every joint
            parent = None
            for j, ets_j in enumerate(arg.split()):
                e_link = ELink(ets_j, parent=parent, name=f"link{j:d}", m=m_list[j], r=r_list[j], I=I_list[j])
                if (e_link.qlim is None
                        and e_link.v is not None
                        and e_link.v.qlim is not None):
                    e_link.qlim = e_link.v.qlim
                parent = e_link
                links.append(e_link)
            # calling ERobot constructor
            super().__init__(links, **kwargs)
        elif islistof(arg, ELink):
            links = arg
            super().__init__(links, **kwargs)
        else:
[... 60 lines cut ...]
                    v[j] = vJ
                    a[j] = Xup[j] * a_grav
                else:
                    jp = robot[j].parent.jindex
                    v[j] = Xup[j] * v[jp] + vJ
                    a[j] = Xup[j] * a[jp] + v[j] @ vJ

            f[j] = I[j] * a[j] + v[j] @ (I[j] * v[j])

        # backward recursion
        n_joints = robot.n - 1
        for j in reversed(range(0, n)):
            # only necessary to compute joint moment if robot has joint at that link
            if robot[j].isjoint:
                # next line could be np.dot(), but fails for symbolic arguments
                Q[n_joints] = np.sum(f[j].A * s[j])
                n_joints -= 1

            if robot[j].parent is not None:
                jp = robot[j].parent.jindex
                f[jp] = f[jp] + Xup[j] * f[j]

        return Q

```

jenskober, 2022-03-24:
starting from Jesse's fix, here is my attempt at modifying the rne function. It now runs through without any errors for a number of URDF files I tested, but the resulting torque is quite different from an equivalent model using a DH definition. Hope that helps somebody a bit to finally get this working.

```py
    # inverse dynamics (recursive Newton-Euler) using spatial vector notation
    def rne(robot, q, qd, qdd, symbolic=False, gravity=None):

        n = robot.n

        # allocate intermediate variables
        Xup = SE3.Alloc(n)
        Xtree = SE3.Alloc(n)

        v = SpatialVelocity.Alloc(n)
        a = SpatialAcceleration.Alloc(n)
        f = SpatialForce.Alloc(n)
        I = SpatialInertia.Alloc(n)  # noqa
        s = []  # joint motion subspace
        if symbolic:
            Q = np.empty((n,), dtype="O")  # joint torque/force
        else:
            Q = np.empty((n,))  # joint torque/force

        # A temp variable to handle static joints
        Ts = SE3(np.eye(4, dtype="O"), check=False)

        # A counter through joints
        j = 0

        # initialize intermediate variables
        for link in robot.links:
            if link.isjoint:
                I[j] = SpatialInertia(m=link.m, r=link.r)
                if symbolic and link.Ts is None:
                    Xtree[j] = SE3(np.eye(4, dtype="O"), check=False)
                else:
                    Xtree[j] = Ts * SE3(link.Ts, check=False)

                if link.v is not None:
                    s.append(link.v.s)

                # Increment the joint counter
                j += 1

                # Reset the Ts tracker
                Ts = SE3(np.eye(4, dtype="O"), check=False)
            else:
                Ts *= SE3(link.Ts, check=False)

        if gravity is None:
            a_grav = -SpatialAcceleration(robot.gravity)
        else:
            a_grav = -SpatialAcceleration(gravity)

        # forward recursion

        # A counter through joints
        j = 0

        for link in robot.links:
            if link.isjoint:
                vJ = SpatialVelocity(s[j] * qd[j])

                # transform from parent(j) to j
                Xup[j] = link.A(q[j]).inv()

                jointParent = link.parent
                while not (jointParent is None or jointParent.isjoint):
                    jointParent = jointParent.parent

                if jointParent is None:
                    v[j] = vJ
                    a[j] = Xup[j] * a_grav + SpatialAcceleration(s[j] * qdd[j])
                else:
                    jp = jointParent.jindex
                    v[j] = Xup[j] * v[jp] + vJ
                    a[j] = Xup[j] * a[jp] + SpatialAcceleration(s[j] * qdd[j]) + v[j] @ vJ

                f[j] = I[j] * a[j] + v[j] @ (I[j] * v[j])

                # Increment the joint counter
                j += 1

        # backward recursion

        # A counter through joints
        j = n-1

        for link in reversed(robot.links):
            if link.isjoint:
                # next line could be np.dot(), but fails for symbolic arguments
                Q[j] = np.sum(f[j].A * s[j])

                jointParent = link.parent
                while not (jointParent is None or jointParent.isjoint):
                    jointParent = jointParent.parent

                if jointParent is not None:
                    jp = link.jindex
                    f[jp] = f[jp] + Xup[j] * f[j]

                # Decrement the joint counter
                j -= 1

        return Q
```

petercorke, 2026-08-25:
Firstly, I'd like to apologise for taking well over five years to get back to you, but essentially my day job and life got in the way. Over the last few months I've been able to devote a lot more time to this toolbox — fixing bugs, reworking dubious design choices, and developing tests and solid CI processes to prevent future drift.

On this specific one, I owe you a more honest answer than "fixed, please retry": the literal crash you hit is gone, but I traced through it and found the underlying computation is still wrong — it now silently returns near-zero torques instead of crashing, which isn't really progress. Root-caused and tracked properly as #636, which also references a real working patch already proposed in #483. Thank you for the original report — it pointed at a real, still-unresolved bug.

closed 2026-08-25 by petercorke