petercorke/robotics-toolbox-python #240: Swift is not installed, " "install it using pip or conda (FIX)

Jakubach, 2021-05-17:
Hello,
while working with Swift we found an issue, we are getting message:
"Swift is not installed, " "install it using pip or conda".
Problem can be solved by changing line 1244 in robotictoolbox\robot\Robot.py from:
`from roboticstoolbox.backends.swift import Swift`
to:
`from roboticstoolbox.backends.Swift import Swift`

Tested on Windows on below example:

```from spatialmath import *
import roboticstoolbox as rtb
from roboticstoolbox.tools.trajectory import *

robot = rtb.models.Panda()
T = SE3(0.7, 0.2, 0.1) * SE3.OA([0, 1, 0], [0, 0, -1])
solution = robot.ikine_LM(T)
traj = jtraj(robot.qz, solution.q, 50)

# Swift
robot.plot(traj.q, backend = 'swift')```

jhavl, 2021-05-31:
Hey @Jakubach, this may be a legacy problem, as the Swift module is named `swift` with a lower case. What version of the toolbox and swift are you running? This could be caused by incompatible versions.

carlosxsm, 2021-06-04:
I am also having issues with Swift
I am using mybinder.
"Could not connect to the Swift simulator"

jhavl, 2021-06-07:
@carlosxsm, Swift will not work with mybinder due to mybinder operating on a remote server.

Jakubach, 2021-06-10:
> 
> 
> Hey @Jakubach, this may be a legacy problem, as the Swift module is named `swift` with a lower case. What version of the toolbox and swift are you running? This could be caused by incompatible versions.

I was running RTB v 0.10.0 and Swift v 0.9.6

closed 2021-09-13 by jhavl