petercorke/robotics-toolbox-python #243: End-effector parameter missing in `ikine` methods

tsoud, 2021-05-21:
Hello `robotics-toolbox-python` team,

First off, I have to say it's wonderful that you brought this toolbox to Python, and even better, that this project is led by the creator of the original Matlab version! Big fan of Professor Corke and his work.

**Describe the bug**
I'm currently working on  a model of a Kinova Gen3 to compute forward and inverse kinematics to pass into a Webots simulation. This arm has multiple end-effector options which requires passing an argument specifying the end-effector type to the solver. There is a parameter `end` to pass this argument to the `fkine` method but that parameter is absent for `ikine_XX` methods.

**Environment**
- Your OS (MacOS, Linux, Windows): Linux (Mint 20.1)
- Your Python version: 3.8.5
**Version information**
Did you install from PyPI or GitHub? Conda install from conda-forge repo
What version number? roboticstoolbox-python 0.10.1

**To Reproduce**
Load a model of a manipulator with multiple end-effector options (e.g. Kinova Gen 3) and select an end-effector:
```
kvG3 = rtb.models.URDF.KinovaGen3()
end = kvG3.ee_links[0].name  # 'camera_link'
```
Calculate an end-pose transform, then use `ikine_XX` to calculate joint positions:
```
Tr = kvG3.fkine(kvG3.qr, end=end)
IK_sol = kvG3.ikine_LM(Tr, end=end)
```
Python raises a `ValueError` that an end-effector must be specified:
[image]

Note how `fkine` (above) takes an `end` argument to specify this but `ikine_XX` methods do not. There does not seem to be a way to add an end-effector option:
```
IK_sol = kvG3.ikine_LM(Tr, end=end)
TypeError: ikine_LM() got an unexpected keyword argument 'end'
```
[image]
```
import inspect
print(inspect.getfullargspec(kvG3.ikine_LM).args)
```
```
['self', 'T', 'q0', 'mask', 'ilimit', 'rlimit', 'tol', 'L', 'Lmin', 'search', 'slimit', 'transpose']
```

**Expected behavior**
`ikine_XX` methods take an additional argument `end` to specify the end-effector. Perhaps better would be to make it an attribute of the robot class with a default value for arms with multiple end-effector options.

petercorke, 2021-05-21:
Thanks for kind words.  What exactly do you mean by EE options for this robot?  Do you want to select a specific link frame or add a specific SE3 to the end of the robot?

This is a bit of the toolbox that hasn't settled down properly and we've explored a few ideas.  Be happy for your input here.

tsoud, 2021-05-22:
Hello Peter,

I don't really need an EE, but neither `fkine()` nor `ikine_XX()` will return a solution for the `KinovaGen3` robot model (which has multiple end-effectors) unless an EE is specified. Instead, they will raise `ValueError: Must specify which end-effector`. Here is the note from the docstring for `fkine()`:
```
:param end: end-effector or gripper to compute forward kinematics to :type end: str or ELink or Gripper
...
Note

- For a robot with a single end-effector there is no need to specify `end`
- For a robot with multiple end-effectors, the `end` must be specified.
- The robot's base tool transform, if set, is incorporated into the result.
- A tool transform, if provided, is incorporated into the result.
- Works from the end-effector link to the base
```
With `fkine()` the solution is to pass an argument to the `end` parameter. For example, passing `end='end_effector_link'` mimics a robot without an EE. 

`ikine()` does not have an `end` parameter, but will return the same `ValueError: Must specify which end-effector` for this robot. Basically, I can't get a result from `ikine()` because of this error. 

For now, I will try to work around the error by creating a custom `ERobot` instance using the information from the KinovaGen3 URDF model.

tsoud, 2021-05-23:
@petercorke, @jhavl, 
After digging a bit deeper into this, I see that the IK solvers make calls to `fkine()` to update the cost value. As a quick fix, I manually added the `end` parameter (with a default argument of `None`) to each function in `IK.py` and passed that argument wherever there is a call to `fkine()`, `jacobe()` or `jacob0()`. 

So far, it looks like it's working:
```
>>> kvG3 = rtb.models.URDF.KinovaGen3()
>>> end = kvG3.ee_links[0].name  # 'camera_link'
>>> Tr = kvG3.fkine(kvG3.qr, end=end)
>>> IK_sol = kvG3.ikine_LMS(Tr, end=end) 

IKsolution(q=array([-0.31516199,  0.30520632,  0.26651699,  1.60723415, -0.07430546,
        0.99893762, -1.60814182]), success=True, reason=None, iterations=10, residual=4.623872493222165e-13)
```

petercorke, 2021-05-26:
This is a good solution and should be added to the codebase.  The idea of a default EE is also a good one.  We are rather inconsistent about tool transforms, `fkine` has it, and therefore so should `ikine`, but it does at least default to the attribute of the Robot class.

tsoud, 2021-05-26:
> This is a good solution and should be added to the codebase. The idea of a default EE is also a good one. We are rather inconsistent about tool transforms, `fkine` has it, and therefore so should `ikine`, but it does at least default to the attribute of the Robot class.

If you would like me to create a PR for the "quick fix", just let me know.

danielj195, 2021-07-13:
I ran into the same issue.  Can a PR be made for this change?

jhavl, 2021-09-13:
Should be fixed now, let me know if problems persist

closed 2021-09-13 by jhavl