petercorke/robotics-toolbox-python #244: launching non-default firefox browser with swift not working ... (How) CAN I open tkinterweb with swift?

Giessen, 2021-05-22:
Env:
---
Windows 10, python3.7.9

Version:
---
robotics-toolbox-python
branch Master
commit 338d960

Reproduce:
---
import roboticstoolbox as rtb
#robot = rtb.models.DH.Panda()
robot = rtb.models.URDF.Panda()
from roboticstoolbox.backends.swift import Swift  # instantiate 3D browser-based visualizer

backend = Swift()
backend.launch(browser='Firefox')            # activate it
backend.add(robot)    # add robot to the 3D scene
=>
Could not open specified browser, using default instead.

Question:
---
(How) CAN I open tkinterweb (or chromium) with swift?
https://github.com/Andereoo/TkinterWeb

mfkenson, 2021-05-31:
@Giessen the underlying magic is done by webbrowser python package. You may want to look into that if you want it tkinterweb/chromium support. [here](https://github.com/jhavl/swift/blob/2a0c93e604089f0c8086a03d19521b4d15169a8f/swift/SwiftRoute.py#L48) is where swift open the browser 

I just tried to run your code. Successful with both 'google-chrome' and 'firefox' on Python3.6; Ubuntu 18.04. latest commit of robotics-toolbox-python (338d960) and swift (2a0c93e).

```
backend.launch(browser='firefox')
```

```
backend.launch(browser='google-chrome')
```

jhavl, 2021-06-07:
@Giessen if the `browser` keyword argument is not present in the `launch` method (as shown by @mfkenson) then swift will open in the default browser. Opening in another browser is attempted to be handled by the webbrowser python package. However, their keywords for choosing a browser varies depending on the platform with many browsers not supported.

See [this link](https://docs.python.org/3/library/webbrowser.html#webbrowser.register) for a list of supported browsers. Use the `Type Name` from the table with  `backend.launch(browser='...')` to launch the specific browser.

closed 2021-06-07 by jhavl