petercorke/robotics-toolbox-python #245: Add a fourth DH parameter "offset" to the README in the DH models folder.

MCollins52, 2021-05-24:
Standard DH tables include 4 parameters while the README in the DH models folder only mentions 3 of the 4. The missing parameter is Θ, which is the link twist offset (or joint offset). The README should be updated to something like the following to make clear that all 4 parameters can be used within the toolbox. This is also within the __init__ section of the README.

  offset=0,       # link twist offset (Dennavit-Hartenberg notation)
  d=0,              # link length (Dennavit-Hartenberg notation)
  a=0,              # link offset (Dennavit-Hartenberg notation)
  alpha=pi/2,   # link twist (Dennavit-Hartenberg notation)

petercorke, 2021-05-26:
You're going to have to be clearer.

- What do you mean by *init* section of the README?
- There are 4 DH params: theta, d, a, alpha and these are always displayed. Offset is not really a DH parameter, but is shown in either the theta or d column as appropriate.

jhavl, 2021-09-13:
Since offset is not a DH paramter, it shouldn't be displayed in the table. However, it can be easily visualised with `print(robot.offset)`.

closed 2021-09-13 by jhavl