petercorke/robotics-toolbox-python #248: fkine gives incorrect value when base_transform is to be excluded

suddrey-qut, 2021-05-27:
**Describe the bug**
The fkine method always applies the base transform to the result, regardless of whether start_link is specified or include_base=False.

**Version information**

Installed: PyPi
Version: 0.10.1

**Expected behavior**
The expected behaviour would be that if start_link=robot.base_link the base transform is excluded, likewise if include_base=False then the base transform is excluded.

petercorke, 2021-05-30:
Do you actually use start link?  I'm not sure anymore what the use case is.

jhavl, 2021-05-30:
fkine, and several other kinematic methods feature `start` and `end` keywords to specify the reference frames for the associated method.

So if you wanted the transform from the elbow to the end effector, you can say `fkine(q, start=elbow, end=ee)`.

If you want the forward kinematics of the robot in the robot base frame, rather than the world frame, you can do either `fkine(q, start=base_link)` or `fkine(q, include_base=False)`. However, both still include the transform of the robot base in the world frame which is incorrect.

jhavl, 2021-05-30:
This also means that the `include_base` keyword argument is redundant and should be removed. Any thoughts @petercorke?

petercorke, 2021-05-31:
Let's not rush this, but clearly base should not be included in Gav's case. `include_base` may have some merit. 

Given the limits of start/end we've discussed, they have to be on same branch, and rarity of use case, people could get the result they want by
```
FK(end=L1).inv() * FK(end=L2)
```
Which is not a huge hardship.

commit 189186c4 (2021-06-07): Fixes #248, base included with defined start frame
closed 2021-06-07 by jhavl