petercorke/robotics-toolbox-python #253: Joint axis restricted to one of the frame axis

gianluca-garofalo, 2021-06-29:
While using Swift, I have noticed that a URDF file containing a joint axis that is not aligned with one of the three frame axis (either x, y or z) gets the wrong number of DoFs. Although I noticed this while using Swift, I suspect that it might be due to this part of the code:
https://github.com/petercorke/robotics-toolbox-python/blob/338d960a220b4a778278344e763d2b31b247f47b/roboticstoolbox/tools/urdf/urdf.py#L1740

If I am correct, do you have a suggestion on how to fix this issue that does not involve changing the URDF?
Thank you in advance for your support and of course for sharing your code!

jhavl, 2021-07-19:
Hey @gianluca-garofalo could you please provide an example of a URDF which produces this issue? Then I can have a look at what we can do about it.

gianluca-garofalo, 2021-07-19:
Hi @jhavl,

Using just one joint should already be enough. I have attached a minimal example, that I hope shows what I was referring to. If you change the axis to "1 0 0", the test is visualized correctly, while the axis in the attached URDF yields an error.
My guess is that the URDF parser does not recognize an axis that is not x, y or z; that is why I pointed to the code line in my previous message.

Please, do let me know if I can support you somehow.

Thank you for your looking into it.
Gianluca

[issue253.zip](https://github.com/petercorke/robotics-toolbox-python/files/6838879/issue253.zip)

jhavl, 2021-09-13:
Sorry for the delay, I've implemented a fix, let me know if it works as expected.

closed 2021-09-13 by jhavl

wildthingz, 2021-09-17:
Hello,

I'd like to re-open this issue. I don't think this is resolved.

While moving a revolute joint with a tilted `<axis xyz="0 0.866025 -0.5"/>` for instance I only see movement in one direction (i.e. z driection).

Cheers,

Hatef

hubble14567, 2024-05-20:
Hello,

I discovered this closed issue after I made my pull request. The issue is still here in the current version. You can find my (quite long) fix in this pull request:
#441

I hope it will help someone.

commit c05f4cad (2021-09-13): Fix #253