petercorke/robotics-toolbox-python #257: Pyplot not plotting DH robot

olmerg, 2021-08-24:
**Check here first**
[Common issues](https://github.com/petercorke/robotics-toolbox-python/wiki/Common-Issues)

**Describe the bug**
The pyplot in inline mode (spyder or jupyter notebook) of DH robot do not show the robot, I think that is the last update of some dependency.

**Version information**

Did you install from PyPI or GitHub?
I install the last version with pip

Robotics Toolbox depends heavily on two other packages: Swift (3D graphics) and SpatialMath toolbox (underpinning maths utilities).  If you think your issue is related to these, then please answer the questions above for them.

**To Reproduce**
```
import matplotlib.pyplot as plt
import roboticstoolbox as rtb
%matplotlib inline  
puma = rtb.models.DH.Puma560()
puma.q=puma.qz
puma.plot(puma.qz, block=False)
```
generate a 3dplot without the robot draw

**Expected behavior**
See the robot like happen if the back is qt5 or before the error.

**Screenshots**

**Environment (please complete the following information):**
 - WINDOWS
- Python 3.8.5.
- anaconda

jhavl, 2021-09-13:
Hey @olmerg, do you know roughly the date when this was last working?

rener2020, 2021-09-13:
sciview can't display the image normally

Reqin, 2021-09-13:
bug in ubuntu too

olmerg, 2021-09-13:
I do not know the exact date, because I do not use the library in June and July ,  I have tried to downgrade the matplotlib without success.
My student starts to install the library after 10 de august because this class I teach spatialMath(working great).

olmerg, 2022-02-10:
for me only work with Qt backend in a notebook with vscode
`%matplotlib Qt`

core-engineering, 2022-02-25:
Hi,

A workaround for me is to use `%matplotlib widgets` in jupyterLab and in vscode

Camille

jhavl, 2022-04-07:
Thanks for the workarounds everybody, I've added a section to the common issues section on the wiki to help future people.

closed 2022-04-07 by jhavl