petercorke/robotics-toolbox-python #265: sciview can't display the image normally

KR-serenity, 2021-09-11:
Hi,guys,when i run this code in my pycharm,i can't see the image in my sciview,but others can.It confuse me for three days.Pls help me!
this is the code.

-----------------------------------------
import roboticstoolbox as rtb

from spatialmath import SE3
robot = rtb.models.DH.Panda()
print(robot)

T = robot.fkine(robot.qz)  # forward kinematics
print(T)

T = SE3(0.7, 0.2, 0.1) * SE3.OA([0, 1, 0], [0, 0, -1])
sol = robot.ikine_LM(T)         # solve IK
print(sol)
robot = rtb.models.DH.Panda()
print(robot)
T = robot.fkine(robot.qz)  # forward kinematics
print(T)

T = SE3(0.7, 0.2, 0.1) * SE3.OA([0, 1, 0], [0, 0, -1])
sol = robot.ikine_LM(T)
# solve IK
print(sol)

q_pickup = sol.q
print(robot.fkine(q_pickup))
# FK shows that desired end-effector pose was achieved

qt = rtb.jtraj(robot.qz, q_pickup, 50)
robot.plot(qt.q, movie='panda1.gif')
--------------------------------------------------------------
I uplord the photo which shows my pycharm

[image]

jhavl, 2021-09-13:
This seems to be related to #257

jhavl, 2022-04-07:
Hi @KR-serenity, have a go at the workaround shown in #257, or checkout here https://github.com/petercorke/robotics-toolbox-python/wiki/Common-Issues#visualising-robots-using-matplotlib-in-jupyter-or-sciview

closed 2022-04-07 by jhavl