petercorke/robotics-toolbox-python #266: Acceleration Calculation of the RNE in modified DH

hanm2019, 2021-09-13:
**Describe the bug**
the `vd` is [code](https://github.com/petercorke/robotics-toolbox-python/blob/4d838d5e8a826513ec1c1fad14408f1e83bd68aa/roboticstoolbox/robot/DHRobot.py#L1468) `vd_ = Rt @ _cross(wd, pstar) + _cross(w, _cross(w, pstar)) + vd` may be have some problem.

**Version information**
master

**Expected behavior**
when the joint is revolute, I think the `vd` = `Rt@(vd + _cross(wd,pstar) + _cross(w, _cross(w, pstar)))`

since, 
[image]
then, We add the RT trans at each side.
[image]

hanm2019, 2021-09-13:
so as the prismatic axis

hanm2019, 2021-09-13:
I read the ne.c [code](https://github.com/petercorke/robotics-toolbox-python/blob/4d838d5e8a826513ec1c1fad14408f1e83bd68aa/roboticstoolbox/core/ne.c#L179), It also show that, after calculate the sum of  a_i, cross(wd,pstar), cross(w,cross(w,pstar)), mult a ROT to transform the reference  axis

hanm2019, 2021-09-13:
I write a test:
```
import numpy.testing as nt
import numpy as np
import roboticstoolbox as rp
import spatialmath as sm
import unittest
import math

panda = rp.models.DH.Panda()
print(panda.mdh)

dq = np.array([1,1,1,1,1,1,1])
ddq = np.array([1,2,3,4,5,6,7])
t0 = panda.rne(panda.qr, dq, ddq)
tp0 = panda.rne_python(panda.qz, dq, ddq, debug=True)
print('t0:',t0)
print('tp0:',tp0)
```
but get two difference result. 
since the Panda Robot'm is unset, default is 0, so, I change the Link.py:143 to `dynchange += dynpar(self, "m", m, 1.0)` so, each Link is 1 kg

the testcase get two result:
```
t0: [  3.69972741 -12.09275043   4.30749236  13.13058012   0.22469786   1.01257754   0.        ]
tp0: [ 1.09406   -0.20341    1.09406    0.9699845  1.04192    0.864377   0.       ]
```
so, I think the rne and rne_python may have some bug in it. 

Even I update the vd_ in Line [1468](https://github.com/hanm2019/robotics-toolbox-python/blob/0692d664721bde2bf468380de76a2f71f769b633/roboticstoolbox/robot/DHRobot.py#L1468), the result also not same.

After read the rne_python code, I think the Line[1539](https://github.com/petercorke/robotics-toolbox-python/blob/4d838d5e8a826513ec1c1fad14408f1e83bd68aa/roboticstoolbox/robot/DHRobot.py#L1539) maybe have some problem.
 ```
nn_ = (
                        R @ nn
                        + _cross(pstar, R @ f)
                        + _cross(pstar, Fm[:, j])
                        + Nm[:, j]
                    )
```

nn should be the sum of 4 part of vector, the third part which should be the `link.r x Fm` , not `pstar x Fm` , since the Fm is in the center of mess, not at the endpoint.

hanm2019, 2021-09-13:
by the way, Even I update the nn_, the result of `rne_python` still not same with `rne` , maybe have some other bug :)

petercorke, 2021-11-01:
Quick question, it's been ages since I looked at this code, the `vd` fix is for an inconsistency between the C version, the Python version.  That's not good!  What reference or text are you using for truth?

petercorke, 2026-08-25:
Firstly, I'd like to apologise for taking close to five years to get back to you, but essentially my day job and life got in the way. Over the last few months I've been able to devote a lot more time to this toolbox — fixing bugs, reworking dubious design choices, and developing tests and solid CI processes to prevent future drift.

On this specific one: you were right — the missing parentheses in that `vd_` line were a real bug. I checked the current source directly: it now reads `Rt @ (_cross(wd, pstar) + _cross(w, _cross(w, pstar)) + vd)`, matching your analysis. I also verified it numerically on the UR5 (an MDH model): `RNE - (M·qdd + C·qd + g)` is now ~1e-15 (machine epsilon) where it previously diverged.

Thank you for the precise diagnosis — this was a real, confirmed fix.

closed 2026-08-25 by petercorke