petercorke/robotics-toolbox-python #270: rtb.models.list(mtype="URDF") has some robot with errors

olmerg, 2021-10-05:
**Describe the bug**

The configuration of Frankie robot are bad, it require 10 DOF but only has 9. (lines 54 and 56 of site-packages\roboticstoolbox\models\URDF\Frankie.py)

After solve this problem another problem is presented with one model which the xacro file is not found:

XacroException: No such file or directory: C:\Users\olmer\anaconda3\Lib\site-packages\rtbdata\xacro\kuka_description\kuka_lbr_iiwa\urdf\lbr_iiwa_14_r820.xacro [Errno 2] No such file or directory: 'C:\\Users\\olmer\\anaconda3\\Lib\\site-packages\\rtbdata\\xacro\\kuka_description\\kuka_lbr_iiwa\\urdf\\lbr_iiwa_14_r820.xacro'

if I remove the LBR model now the error is:

XacroException: No such file or directory: C:\Users\olmer\anaconda3\Lib\site-packages\rtbdata\xacro\kortex_description\robots\gen3.xacro [Errno 2] No such file or directory: 'C:\\Users\\olmer\\anaconda3\\Lib\\site-packages\\rtbdata\\xacro\\kortex_description\\robots\\gen3.xacro'

Conclusion, bad link to xacro files:
- LBR
- kinovaGen3 
- yumi
bad configuration definition Frankie(lines 54 and 56 of site-packages\roboticstoolbox\models\URDF\Frankie.py)

**Version information**

Did you install from PyPI or GitHub?

installed with pip version 0.11  over windows with o without anaconda.

**To Reproduce**

import roboticstoolbox as rtb
rtb.models.list(mtype="URDF")

**Expected behavior**
show the list of available robots

**Screenshots**

XacroException: No such file or directory: C:\Users\olmer\anaconda3\Lib\site-packages\rtbdata\xacro\kuka_description\kuka_lbr_iiwa\urdf\lbr_iiwa_14_r820.xacro [Errno 2] No such file or directory: 'C:\\Users\\olmer\\anaconda3\\Lib\\site-packages\\rtbdata\\xacro\\kuka_description\\kuka_lbr_iiwa\\urdf\\lbr_iiwa_14_r820.xacro'

**Environment (please complete the following information):**
 - Your OS  Windows)
- Your Python version 3.8.5

**Additional context**
Add any other context about the problem here.

olmerg, 2021-10-05:
Partial solution
`pip install --upgrade rtb-data`

It will add the xacro files that are not found, I think that in the moment to update roboticstoolbox the rtb-data package is not update.
The frankie description continue failing(lines 54 and 56 of site-packages\roboticstoolbox\models\URDF\Frankie.py)

jhavl, 2022-04-07:
Hey @olmerg, thanks for bringing this to our attention. This issue has been fixed on the future branch and will be incorporated into the next release. Also note that `mtype` has been changed to `type` so correct usage in `rtb.models.list(type="URDF")`.

closed 2022-04-07 by jhavl