petercorke/robotics-toolbox-python #279: issue with swift

sulaiman-01, 2021-11-25:
I am using a UR10 robot and plotting it using swift 
this the code i am using:

`import roboticstoolbox as rtb

model = rtb.models.URDF.UR10()
print(model)
qz = [0,30,-30,0,0,0]
model.plot(qz,backend = 'swift')`

and i get this error:

Exception happened during processing of request from ('127.0.0.1', 37848)
Traceback (most recent call last):
  File "/usr/lib/python3.8/socketserver.py", line 316, in _handle_request_noblock
    self.process_request(request, client_address)
  File "/usr/lib/python3.8/socketserver.py", line 347, in process_request
    self.finish_request(request, client_address)
  File "/usr/lib/python3.8/socketserver.py", line 360, in finish_request
    self.RequestHandlerClass(request, client_address, self)
  File "/home/ash/.local/lib/python3.8/site-packages/swift/SwiftRoute.py", line 221, in __init__
    super(MyHttpRequestHandler, self).__init__(
  File "/usr/lib/python3.8/http/server.py", line 647, in __init__
    super().__init__(*args, **kwargs)
  File "/usr/lib/python3.8/socketserver.py", line 747, in __init__
    self.handle()
  File "/usr/lib/python3.8/http/server.py", line 427, in handle
    self.handle_one_request()
  File "/usr/lib/python3.8/http/server.py", line 415, in handle_one_request
    method()
  File "/home/ash/.local/lib/python3.8/site-packages/swift/SwiftRoute.py", line 263, in do_GET
    self.send_file_via_real_path()
  File "/home/ash/.local/lib/python3.8/site-packages/swift/SwiftRoute.py", line 286, in send_file_via_real_path
    self.copyfile(f, self.wfile)
  File "/usr/lib/python3.8/http/server.py", line 853, in copyfile
    shutil.copyfileobj(source, outputfile)
  File "/usr/lib/python3.8/shutil.py", line 208, in copyfileobj
    fdst_write(buf)
  File "/usr/lib/python3.8/socketserver.py", line 826, in write
    self._sock.sendall(b)
BrokenPipeError: [Errno 32] Broken pipe

and the visualizer still open in the web and even if i remove the "backend = 'swift'" command it still plots it in webswift  instead of using matplolib and gives the same error

`import roboticstoolbox as rtb

model = rtb.models.URDF.UR10()
print(model)
q = [0,30,-30,0,0,0]
model.plot(q)`

sulaiman-01, 2021-11-25:
it is only happening for UR10, i have tried it with Pandas and i dont get the error messege, but it still plots in swift even if i dont mention backend = "swift"

tomawang, 2022-01-28:
Hello:
Please help ! Thanks!  Sincerely, Tom Wang
In above scripts, with backend = "swift" or not, I got the following. 
>>> model.plot(qz)
d:\Anaconda3\lib\site-packages\swift\out
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "d:\Anaconda3\lib\site-packages\roboticstoolbox\robot\Robot.py", line 1419, in plot
    env.add(self, readonly=True, **kwargs)
  File "d:\Anaconda3\lib\site-packages\swift\Swift.py", line 367, in add
    show_robot=show_robot, show_collision=show_collision
TypeError: _to_dict() got an unexpected keyword argument 'show_robot'
>>>

jhavl, 2022-04-07:
Hey @sulaiman-01, your code works fine for me using the Gihub version of the toolbox and Swift.

This issue has somethiing to do with the communication between Python and the webbrowser. Make sure you are using the latest version of the toolbox and swift on Python > 3.6. Also try using the latest version of Chrome as the webbrowser. Feel free to re-open the issue if you do this and it still isn't working.

jhavl, 2022-04-07:
@alaflaquiere, I suspect Swift will not work with WSL as Swift requires a browser tab to open and communicate with. I don't think that WSL can open chrome (feel free to correct me if I'm wrong).

closed 2022-04-07 by jhavl

alaflaquiere, 2022-04-11:
I think I managed to make it work with Chrome, however it didn't work with Firefox. I'm not 100% sure as it was a few weeks back, and I simply switched to Windows since.