petercorke/robotics-toolbox-python #281: error when plotting Planar_Y robot

durka, 2021-12-27:
**Describe the bug**
When attempting to make a noodle plot of the `ETS.Planar_Y` model, it partially plots, but encounters an error and does not show the whole thing (only one link I believe).

**Version information**

- roboticstoolbox 0.11.0 installed from PyPI
- spatialmath 0.11 installed from PyPI

**To Reproduce**
Steps to reproduce the behavior:
1. The shortest, complete, Python script that exhibits the bug.
  ```python
  import roboticstoolbox as rtb
  r = rtb.models.ETS.Planar_Y()
  r.plot(r.qz)
  ```
2. The script output, including error messages.
  ```
  Traceback (most recent call last):
    File "<stdin>", line 1, in <module>
    File "/home/alex/.cache/pypoetry/virtualenvs/sim-bd0j9NSV-py3.9/lib/python3.9/site-packages/roboticstoolbox/robot/Robot.py", line 1442, in plot
      env.step(dt)
    File "/home/alex/.cache/pypoetry/virtualenvs/sim-bd0j9NSV-py3.9/lib/python3.9/site-packages/roboticstoolbox/backends/PyPlot/PyPlot.py", line 203, in step
      robot.draw()
    File "/home/alex/.cache/pypoetry/virtualenvs/sim-bd0j9NSV-py3.9/lib/python3.9/site-packages/roboticstoolbox/backends/PyPlot/RobotPlot.py", line 106, in draw
      self.links[0].set_3d_properties(points[:, 2])
    File "/home/alex/.cache/pypoetry/virtualenvs/sim-bd0j9NSV-py3.9/lib/python3.9/site-packages/mpl_toolkits/mplot3d/art3d.py", line 175, in set_3d_properties
      zs = np.broadcast_to(zs, len(xs))
    File "<__array_function__ internals>", line 5, in broadcast_to
    File "/home/alex/.cache/pypoetry/virtualenvs/sim-bd0j9NSV-py3.9/lib/python3.9/site-packages/numpy/lib/stride_tricks.py", line 411, in broadcast_to
      return _broadcast_to(array, shape, subok=subok, readonly=True)
    File "/home/alex/.cache/pypoetry/virtualenvs/sim-bd0j9NSV-py3.9/lib/python3.9/site-packages/numpy/lib/stride_tricks.py", line 348, in _broadcast_to
      it = np.nditer(
  ValueError: operands could not be broadcast together with remapped shapes [original->remapped]: (4,)  and requested shape (3,)
  ```
[image]

**Expected behavior**
I was really hoping this would be a simple way to make noodle plots for branched kinematic trees (without having to write a giant URDF nightmare with 3D meshes etc), so I'll be disappointed if the root cause of this is that it simply isn't really supported.

**Environment (please complete the following information):**
- Ubuntu 20.04
- Python 3.9

petercorke, 2021-12-27:
Hopefully something simple, it worked a couple of months back, will check it out in the next day or so.

jhavl, 2022-04-07:
This has been fixed on the future branch, will be incorporated into the next release

commit 63cc7e0c (2022-04-07): Fix #281, fix segment drawing
closed 2022-04-07 by jhavl