petercorke/robotics-toolbox-python #282: Swift issue

hibabfht, 2021-12-28:
**Describe the bug**
I tried the first example of readme, very thing works well until the swift part, the interface shows only axes but not the robot.

**Version information**

Did you install from PyPI
pip version 21.2.4
I installed also swift, sympy, spatialmath-python manually.

Robotics Toolbox depends heavily on two other packages: Swift (3D graphics) and SpatialMath toolbox (underpinning maths utilities).  If you think your issue is related to these, then please answer the questions above for them.

**To Reproduce**
Steps to reproduce the behavior:
1. The shortest, complete, Python script that exhibits the bug.

[code.txt](https://github.com/petercorke/robotics-toolbox-python/files/7785864/t.txt)

2. The script output, including error messages.

[output.txt](https://github.com/petercorke/robotics-toolbox-python/files/7785870/New.Text.Document.txt)

**Expected behavior**
Work as expected, but when it comes to the 3D representation, I expected the robot to appear and move.

**Screenshots**
If applicable, add screenshots to help explain your problem.

[image]

**Environment (please complete the following information):**
 - Windows 10
- Python 3.9.8 64-bits

**Additional context**
/

tomawang, 2022-01-27:
Hello:
I have the same situation, no robot appear ! Please help! Thanks a lot ! 
Sincerely, Tom Wang

### PC:
Default Edge browser, win 10 64-bits,, python 3.7.6 (Anaconda3),
roboticstoolbox-python Version: 0.11.0 (conda install).

### Test code: 
>>> import roboticstoolbox as rtb
>>> panda = rtb.models.URDF.Panda()
>>> panda.plot(panda.qz, backend="swift")
d:\Anaconda3\lib\site-packages\swift\out
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "d:\Anaconda3\lib\site-packages\roboticstoolbox\robot\Robot.py", line 1419, in plot
    env.add(self, readonly=True, **kwargs)
  File "d:\Anaconda3\lib\site-packages\swift\Swift.py", line 367, in add
    show_robot=show_robot, show_collision=show_collision
TypeError: _to_dict() got an unexpected keyword argument 'show_robot'
>>>

jhavl, 2022-04-07:
Hey @hibabfht, Swift does not currently support DH robots and therefore can not be visualised in a swift scene. Use our URDF models with Swift.

jhavl, 2022-04-07:
@tomawang your issue has been fixed on the future branch and will be incorporated into the the release.

closed 2022-04-07 by jhavl