petercorke/robotics-toolbox-python #288: Function "jacobm" doesn't work

vog55, 2022-01-07:
Hello, 

I have a problem with the subfunction "jacobm". According to your paper "Position-Based Servoing Example on an omnidirectional mobile manipulator" you use the subfunction "jacobm" for the linear component of objective function. To understand the sense/purpose of the subfunction, I broke down their source code. In doing so, I constantly get the following error when executing the "jacobm" (see png file). I can't quite understand this, since it is not a singularity. 

[image]

petercorke, 2022-01-22:
Not sure I ever wrote the paper you mention.

`jacobm` is the Jacobian of the manipulator manipulability function, which doesn't seem relevant to visual servoing or omnidirectional mobile platforms.

jhavl, 2022-01-22:
Hey @vog55 ,
To assist further, can you please provide the code for how you got this error to reproduce?

jhavl, 2022-04-07:
The jacobm function is working fine on the future branch with

```
import roboticstoolbox as rtb

panda = rtb.models.Panda()
panda.q = panda.qr
print(panda.jacobm(panda.q))
```

closed 2022-04-07 by jhavl