petercorke/robotics-toolbox-python #290: UR3, UR5, UR10 kinematics differ between DH and URDF versions

micsc12, 2022-01-12:
**Describe the bug**
fkine() returns different result when DH and URDF version of the same robot. 

**Version information**
Installed from PyPI, version 0.11.0

**To Reproduce**
Steps to reproduce the behavior:
```
ur_dh = rtb.models.DH.UR5()
print(ur_dh.fkine([0,0,0,0,0,0]))

ur_urdf = rtb.models.URDF.UR5()
print(ur_urdf.fkine([0,0,0,0,0,0]))
```
Output:
```
   1         0         0        -0.8173    
   0         0        -1        -0.1915    
   0         1         0        -0.005191  
   0         0         0         1         

   0         1         0         0.8173    
   1         0         0         0.1915    
   0         0        -1        -0.005491  
   0         0         0         1  
```

**Expected behavior**
Expected the output of fkine to be equal. Checked with official UR5 simulator, DH models is the correct one. 

**Environment (please complete the following information):**
Ubuntu 20.04, python 3.8

petercorke, 2022-01-22:
It's the same brand/model of robot, but the definition of the base frame, the zero angle configuration, and even the direction of joint rotation could be different.

This is a big problem in robotics, a "model" is not sufficient without that other information which is rarely provided.

micsc12, 2022-01-24:
Hi @petercorke, thanks for replying and thanks for supplying this immensely useful toolbox. I think i will use it a lot in the upcoming years :) 

Should i try and fix the URDF models so it corresponds to the DH models, or should the models in repo just be taken as a starting point, and tweaked locally for my projects? 

I.e. does it make sense to try and streamline the robot model conventions used internally in RTB?

If you think its not worth the hassle at this point, i will just close this issue again.

petercorke, 2022-02-01:
I think just understand the conventions of the model and work with it.  If you don't like the conventions then hack the model.

I understand the frustration around conventions, but for these models we are simply "republishing" them in the hope they are useful.  We're not endorsing the models.

closed 2022-02-01 by petercorke