petercorke/robotics-toolbox-python #291: Adding Custom URDF Robot Models

pranshumalik14, 2022-01-18:
**Describe the bug**
Adding new URDF models (that refer to stl files) needs a relative path to the XACRO path in rtbdata.

`ValueError: file xacro/stl/braccio_base.stl not found locally or in rtbdata`

Any help on how to do this properly will be highly appreciated. Here's a snapshot of the code we have written:

```python
class Braccio(rtb.ERobot):

    def __init__(self):

        links, name, urdf_string, urdf_filepath = self.URDF_read(
            rtb.rtb_path_to_datafile("../urdf/braccio.urdf.xacro", local=True)
        )

        super().__init__(
            links,
            name=name,
            manufacturer="Arduino",
            gripper_links=links[12],
            urdf_string=urdf_string,
            urdf_filepath=urdf_filepath,
        )
```

We would like to keep the URDF model information local to our project than to move it to our installation folders. The code that parses xacro files seems to look for geometry files relative to the XACRO directory.

**Version information**

Did you install it from PyPI or GitHub?
Latest Github version (pulled) Commit hash: 51aa8bbb3663a7c815f9880d538d61e7c85bc470

**Environment (please complete the following information):**
 - Ubuntu 18.04.
- Python 3.6.9.

pranshumalik14, 2022-01-18:
This was despite setting the top-level directory to the project folder. Seemed like that information was not passed into the geometry parser/loadstr for the URDF.

petercorke, 2022-01-22:
The behaviour we should have is that if an abs path is given it should use that, rather than the package relevant one.

pranshumalik14, 2022-01-22:
Yes absolutely, that should fix it, thank you!

olmerg, 2022-01-31:
I use os.chdir(os.path.dirname(__file__)) to solve the problem about absolute path.

https://github.com/olmerg/rtb_serial_robot/blob/main/planar_3dof/Planar3DOF.py

pranshumalik14, 2022-03-05:
Thanks so much @olmerg! Followed your setup step-by-step and got it to work!

petercorke, 2026-08-25:
Sorry for the very long wait on this one — this has been open since 2022.

I've verified this is genuinely fixed: I built a real reproduction (a local custom URDF outside the installed package, with a mesh referenced by a relative path, matching your original setup) and confirmed it loads correctly from an absolute path with no `os.chdir()` workaround needed and no forced package-relative lookup. URDF/xacro loading has been substantially reworked since this issue was filed — it now uses the `xacrodoc` package to properly resolve absolute paths, relative mesh references, and ROS-style `package://` lookups, rather than forcing everything through the toolbox's own package-relative resolver.

This should also mean @olmerg's workaround (and the underlying need for it) is no longer necessary. Please retest on the current release (roboticstoolbox 1.4.0+) and reopen if you still hit this.

closed 2026-08-25 by petercorke